opencpn-radar-pi/radar_pi #240: Can I conduct research using the radar_pi API independently?

leehuoshi, 2023-07-07:
Hello, thank you for your development which allows us to use such a useful feature. 
I have encountered some problems when parsing code, and I wanted to ask about the concept of radar spoke packets. I hope to analyze Radar_Rawdata, but besides replaying it in the OpenCPN API, I currently can't compile it in other IDEs, so I'm stuck and don't know how to proceed. I am using the Raymarine E70498 radar, could you tell me how to read this program to achieve the function I need? Since there is so much content in the data, I'm not very familiar with C++, and I hope to get some guidance.

keesverruijt, 2023-07-11:
The plugin is written for OpenCPN. There is no way you are going to get it to run otherwise.

IDE support for plugin development is hard -- probably the easiest is to copy the plugin to the OpenCPN plugin directories and run entire OpenCPN in the IDE. I've done this in the past with Xcode.

If you want to do research, "just" compile it, run it under OpenCPN and start modifying it -- try adding a few log messages.

Getting the code to compile can be a slog, easiest is probably Linux Ubuntu 22.04 then run ci/circleci-build-debian.sh. 

Sorry but we can't help you if you don't understand CMake, C++, etc.

leehuoshi, 2023-07-12:
I really appreciate you answering my questions. Originally, I thought I could compile the source code needed for parsing radar point clouds from .cpp code into a dll file, and then use Python for further parsing. Since my system is Windows 10, I've encountered many difficulties in compiling and I'm still not clear about what the problem is. Meanwhile, I'm also spending time learning CMake and C++ in hopes of better understanding the code. I originally thought that reading some subroutines would suffice, but it seems that I oversimplified the problem. I'm extremely grateful for the guidance you've provided. I'll give it a try now.