opencpn-radar-pi/radar_pi #242: Offline Radar Analysis / Replay

rdh-evans, 2023-08-14:
Hi,
I have been reading around your issues and CruisersForum about how to replay Halo radar data through opencpn / radar_pi from pcap files, but I'm having difficulty getting it to work using my own recording (Halo20+) and those provided in radar_pi/examples.
Please excuse my ignorance! I am still learning about manipulating networks.

My current setup is my main host machine (Ubuntu 20.04.5), running opencpn 5.8.4 and Radar Plugin 5.3.4.
I am trying to use a guest VirtualBox (Ubuntu 20.04.5), to run tcpreplay such that it can be viewed in opencpn in my host machine. So far no success, with Wireshark seeing the messages coming through from the VirtualBox, however opencpn remains blind to it.
The VirtualBox machine network adapter is bridged to my local network.
I have confirmed that I can connect between my virtualbox and host via tcp, udp and multicast using netcat and iperf.

1. Do I need to edit the pcap files to make them seen by my host machine? e.g. change the source/destination mac addresses/ips
2. Are there any network config options that are recommended to be applied to my host machine?
3. Do I need to specify any specific data connections in opencpn > Options > Connections dialog?

Any tips with setting this up would be appreciated, or a redirection to an existing how-to.

douwefokkema, 2023-08-14:
Answer on all 3 questions is no.
I tried several times (but not too hard) to run tcpreplay on the same machine or a virtual machine on the same hardware and failed. It seems that rcpreplay takes full control of the interface and needs its own hardware. Best is running it on a extra Linux machine, a Rpi is good enough already. It will work straight way, but take care that everything is on the same subnet. Run tcpreplay with sudo.

rdh-evans, 2023-08-15:
Thanks for the tips.
Yes, tcpreplay man page says you cannot replay pcaps and have the messages viewable to applications on the same machine. Something about how tcpreplay injects the messages below the network stack. I thought a VM might get around this, but maybe not.
Anyhow, running it off a Pi worked fine with halo24-23hrs.pcap and halo20/halo20-ranges.pcap example files.
My own file still doesn't work, neither does halo_all_ranges.pcap example which I initially tried.
It appears I have only recorded from one multicast address so that might be the issue with mine.

rdh-evans, 2023-08-15:
After taking another recording, careful to include all multicast addresses, it works fine. Thanks again. 

For anyone else looking to replay radar through opencpn, my setup was as follows:
Data was recorded using tshark on the computer my Simrad Halo20+ radar is attached to:
`sudo tshark -i eth0 -P -f "dst 236.6.7" -a duration:30 -w <my_radar_log_file>.pcap`
Specifying `-f "dst 236.6.7"` meant I only recorded radar messages, not other network traffic.

For replay, I used a Raspberry Pi 4 model b configured for headless mode and SSH password enabled, plugged into an Ethernet switch on my local network on which my laptop sits running opencpn.

On the pi, installed tcpreplay by `sudo apt install tcpreplay`,
Used `scp -r <src> <dest>` to copy the example .pcap files (and my own) from my laptop, running opencpn.
From the Pi I ran: `sudo tcpreplay -i eth0 <path_to_recording>`. You may also want to run it with `--loop=10` if your recording is short.
That's it! The radar plugin should begin to display the radar image in opencpn, so long as your Pi and your ocpn host are on the same network.