opencpn-radar-pi/radar_pi #244: OCPN crash when enable AIS targets on PPI window.

Hakansv, 2023-08-17:
On Windows: (See also this CF [post](https://www.cruisersforum.com/forums/f134/radar-plugin-v5-0-0-released-216057-52.html#post3812717)
Method:
Show PPI.
Enable "AIS/ARPA on PPI" - crash!
I've tried to debug but don't really understand.
This what I get:
Call stack
>	opencpn.exe!ocpnDC::DrawPolygon(int n, wxPoint * points, int xoffset, int yoffset, float scale, float angle) Line 1173	C++
 	opencpn.exe!AISDrawTarget(AisTargetData * td, ocpnDC & dc, ViewPort & vp, ChartCanvas * cp) Line 1493	C++
 	opencpn.exe!AISDraw(ocpnDC & dc, ViewPort & vp, ChartCanvas * cp) Line 1847	C++
 	opencpn.exe!PlugInAISDrawGL(wxGLCanvas * glcanvas, const PlugIn_ViewPort & vp) Line 7101	C++
 	radar_pi.dll!RadarPlugin::RadarCanvas::Render(wxPaintEvent & evt) Line 659	C++
 	[External Code]	
 	[Frames below may be incorrect and/or missing, no symbols loaded for wxbase32u_vc14x.dll]	
 	opencpn.exe!WinMain(HINSTANCE__ * hInstance, HINSTANCE__ * hPrevInstance, char * lpCmdLine, int nCmdShow) Line 930	C++

[image]

bdbcat, 2023-08-18:
Hakan...
The cause of the fault is that "line_shader" is NULL.
If you can debug this in action, please put a breakpoint at ocpndc.cpp : line 1162.  Then please report which path is taken in the next two or three lines, where "line_shader" is established.

Hakansv, 2023-08-18:
Dave..
"Line shader" is not initialized at 1163 since "m_glchartCanvas" is null. See pict where I've followed since RadarCanvas.ccp:659 as of the call stack above.

[image]

bdbcat, 2023-08-18:
Hakan...
Moving upstream, have a look at pluginmanager.cpp : line 7773
void PlugInAISDrawGL(wxGLCanvas *glcanvas, const PlugIn_ViewPort &vp) {
This function first creates the ocpndc, using the supplied glCanvas parameter. Then he calls the AIS draw method, passing the created ocpndc as a parameter.

The m_glChartCanvas in ocpndc should be initialized from this call, the first argument.

keesverruijt, 2023-08-20:
Confirmed as a regression in O 5.8.4 where all drawing was converted to GLES, using shaders exclusively.
Unfortunately, the PluginAISDrawGL API used is not capable of rendering to an arbitrary GL canvas.  It can only render to the viewed chart GL canvas.  That is a bug in OCPN.

The dev team is Investigating what our options are here, for now there are two (bad) workarounds: Either disable the AIS-on-PPI option or revert to using O 5.6.2.

keesverruijt, 2023-08-20:
We need to either:

1. Disable this until the PluginAISDrawGL API call is fixed in O 5.10.
2. Not make this call and draw AIS targets ourselves. This is a pain since we must then also create a list of AIS targets in the plugin.
3. Investigate whether we can "improve" our canvas so it is feasible to call the API again.

commit 68ce4993 (2023-08-24): Fix #244: Disable AIS on PPI in 5.8.0 to 5.8.4 due to broken API in OpenCPN
closed 2023-08-24 by keesverruijt