opencpn-radar-pi/radar_pi #246: FURUNO DRS4DL+ recorded wireshark packets for analysis

bc38, 2023-09-02:
Recorded Furuno's radar performance with wireshark
The radar is connected directly to the rg45 connector of the laptop
The radar was controlled in the TimeZero Pro software
Written on Windows 10
The network card is set to IP address: 172.31.3.10

I hope that this data will help implement support for the FURUNO radar.
Radar still available until Monday

https://mega.nz/file/kFRHlYQJ#59mdq1uQ8gVpaofnhgKTGl9im65WK6asP5F17QiEMAQ

keesverruijt, 2023-09-09:
Thanks. I don't think work will be started by the core team yet. We have too much to do elsewhere, and our self-interest in adding this is not high. It would be great if someone would step up and add support for Furuno though, in which case I'm sure this dump is of (some) value.

RobertWatchful, 2023-12-04:
I am happy to start work on Furuno Radar. I have a DRS4DL and TimeZero. I just had a look using wireshark this afternoon at the controlling of the radar and can start, stop and change range from TCP Commands. I receive what I assume to be the spoke data over UDP. I do have some development experience and may be able to start a project to read the spoke data and make a port for the DRS4DL at least, and I also have a DRS-6A-NXT, although this is on my boat, so I would only look at that if I am successful with the smaller radar.

douwefokkema, 2023-12-04:
A good point to start is to add your radar in RadarType.h. Take the last set of includes and add #include "furuno/Furunotype.h"
Above you will have to add all Furuno specific files. Then make all missing files in analogy with the ones for existing radars.
For the real hacking you go into the control and receive files. These steps will add Furuno to the radar selection list in preferences.

RobertWatchful, 2023-12-04:
This is very helpful, Thank you for your advice. I will probably start work over the Christmas period.

basau74, 2024-05-23:
Is there any Progress on this? I want to be able to get scanline pixels from network packets of a Furuno radar and would love to help here. 
I think I am able to differentiate parts that might be the scanlines in each packet, but the data does not make much sense. The scanlines are variable length and contain some recognizable byte patterns:

<details>

<summary>8 Scanlines from a packet</summary>
Copied from packet 5,608 of the `02 TX, STB.pcapng` file.

- Angle(?)\
Data

1. 7c 1a 00 00\
00 d1 14 28 3c 50 11 3c 28 14 00 81 08 10 18 20 11 38 54 6c 88 11 78 6c 5c 50 51 3c 28 14 00 fd 00 fd 00 fd 00 fd 00 85
2. 86 1a 00 00\
fe 14 c6 21 78 6c 5c fe 00 fe 00 fe 00 fe 00 ca
3. 90 1a 00 00\
fe 14 86 21 20 44 64 fe 00 fe 00 fe 00 fe 00 fe 00 0a 00 fe
4. 9a 1a 00 00\
fe 14 fe 50 06 44 38 2c 20 11 18 10 08 fe 00 fe 00 fe 00 fe 00 8a 00 fe
5. a4 1a 00 00\
d6 08 10 18 20 11 18 10 08 a6 14 28 3c 50 11 5c 6c 78 fe 00 fe 00 fe 00 fe 00 ea 00
6. ae 1a 00 00\
fe 08 e6 6c 54 38 20 1d fe 00 fe 00 fe 00 fe 00 aa 00 fe 00
7. b8 1a 00 00\
fe 08 a6 08 10 18 20 11 38 54 6c fe 00 fe 00 fe 00 fe 00 ea
8. c3 1a 00 00\
d6 14 28 3c 50 11 3c 28 14 fe 20 fe 00 fe 00 fe 00 fe 00 ba

Note that the angle information are in big-endian byte order.

</details>

I can imagine that they use some form of compression but I have no experience in this field. Any suggestions which one I should test or which ones are common for radar systems?

sq2dk, 2025-02-25:
I'm cross-posting this information from https://github.com/keesverruijt/mayara/issues/4

I have spent way too much time trying to decode this format, but I have some results, yet still some aspects of the data encoding are not understood to me. Below are two images - first showing slice of the radar picture taken while recording data (  @keesverruijt ), second reconstructed image from network data. Of course, image made from network data is simple rectangular (not polar) representation of spokes, so image is distorted, but it allows to see, that both images are showing corresponding data.
[image] 
[image]
Now, from the second image you can see, that echoes are only outlines - this is one of the problem I have - I do not understand (yet) how echo fill data is defined. Another unsolved problem is the length of the data line - I thought, that it would be fixed length after de-compressing, but it seems, that this might be not the case, or I did not worked out compression algorithm fully. Because of that, data I have been working on had to be hand-divided in the lines.
**How data is compressed** : 
consecutive byte value of the data line = X
- if last bit of the byte value ( X ) is 1 ( x && 0x01 > 0) : this is multiple signal byte it means, that there is X>>2 (2 bit shift right) repetitive signal (max?) bytes in the stream
-  if second to last bit of the byte value ( X ) is 1 ( X && 0x2 > 0) : this is multiple signal byte it means, that there is X>>2 (2 bit shift right) repetitive no signal ( 0 ) bytes in the stream

That's basically it. Other values seem to represent the single pixel values (there seem to be about 16 discrete levels). 
That gives approximately 950 pixels line (3Nm range) and seem to be approx 100pix per Nm resolution. 
There might be some trick to determine if fill line should be filled with zeroes or high values, but I cannot find it - thus I'm getting only outlines of the echoes.
First line in each sequence is coded in some other way and I just ignored it for now.
All this is easy when you now how it works, but figuring it out took me a long time :-) .

I'm attaching also spreadsheet with data packet analysis. Some information in there available on what's where in the header and antenna angle encoding.

[packet analysis.ods](https://github.com/user-attachments/files/18961765/packet.analysis.ods)

Jed SQ2DK

malcolmh, 2025-05-06:
This topic seems to be dead - is anybody working on Furuno DRS radars? I would like to contribute - I have a DRS4D-NXT running on TimeZero Navigator. I have successfully captured network traffic in various states from this setup. My programming skills are C and Java, so C++ is out of my comfort zone! I have MacOS, Linux & Windows11 systems that I can run OpenCPN on for testing.

keesverruijt, 2025-05-06:
Furuno support is being added to [mayara](https://github.com/keesverruijt/mayara/actions), which is charting-application-neutral. 

Once that is in shape, either we gut radar_pi to talk to mayara, or create a new plugin. This will have the benefit of being able to run OpenCPN via wireless, as long as there is a small computer running mayara that is connected to the radar via wired ethernet.

keesverruijt, 2025-05-06:
@malcomh can you DM me (kees.verruijt) on Discord?