opencpn-radar-pi/radar_pi #249: Why radar_pi does not draw doppler when connecting halo A, B?

GASTER2302, 2023-09-21:
I have a halo 20+ radar and radar_pi does not want to draw doppler and do doppler auto track

keesverruijt, 2023-09-21:
Are you sure you have a 20+ and not a 20? 20 does not have Doppler.

Do you see doppler on your chart plotter (if you have one)? In case you enable Doppler there, what happens in OpenCPN?

GASTER2302, 2023-09-21:
I am sure it is halo 20+ with doppler.
I dont see any changes on the chart plotter except for a yellow circle in the center in doppler normal mode

keesverruijt, 2023-09-21:
You know what Doppler does right? It colours targets coming towards you or going away from you in a different color. Targets going from right to left or vice versa are not picked up by doppler.

If you change the doppler mode on the chart plotter the doppler mode in the plugin menu should also change.

Are you using v5.5.0 ?

If so, upload a pcap file of about 10-20 seconds during which you disable/enable the doppler function (preferably on the chart plotter).

keesverruijt, 2023-09-21:
Targets have to move at (by default) 3 or more knots towards or away from you.

GASTER2302, 2023-09-21:
In bird mode same?

keesverruijt, 2023-09-21:
I wouldn't know but I presume so. I have never used bird mode.

GASTER2302, 2023-09-21:
Doppler mode, as I understand it, doesnt extinguish the terrain?

keesverruijt, 2023-09-21:
No.

GASTER2302, 2023-09-21:
And can you send me your settings?

GASTER2302, 2023-09-21:
And what does the doppler threshold setting affect?

GASTER2302, 2023-09-21:
As I see it, this is not a Doppler selection of a moving target, but simply a comparison between the readings of the target's shifts towards or away from the radar.  Because looking at the packets that go from the radar to radar_pi, one gets the impression that the state of the target is just arriving.  Because the package with Doppler must have the same size or more than the package with heatmap.

GASTER2302, 2023-09-21:
Just in case, I ask if you have access to the sdk for configuring the radar?  If there is, then it will be added to the radar.

keesverruijt, 2023-09-21:
> As I see it, this is not a Doppler selection of a moving target, but simply a comparison between the readings of the target's shifts towards or away from the radar. Because looking at the packets [...]

The amount of data sent by the radar does not change. But without doppler mode, the heat map for each pixel is 0..15 (4 bits, 1 nibble). With normal Doppler mode values 14 and 15 become special values meaning "receding" and "approaching". Thus any Doppler shift detected by the radar is mapped onto these special values; it is "true Doppler". Thus there are fewer values remaining for the heat map for static or traversing targets. With "Approaching only" only 15 becomes a special value.

> Just in case, I ask if you have access to the sdk for configuring the radar?

No.

> If there is, then it will be added to the radar.

I don't understand this sentence.

keesverruijt, 2024-10-02:
No activity, closing.

closed 2024-10-02 by keesverruijt