opencpn-radar-pi/radar_pi #251: OpenCPN always crashes when using radar_pi for 2 minutes to over an hour

ERousseletChorus, 2023-10-30:
Hi,

I've been using OpenCPN and the plugin Radar Pi to detect ARPA and generate xxTTM messages for months with a Halo20 radar.
To this day, OpenCPN ended up crashing every single time after 2 minutes to over an hour of normal functionning, without me interacting with it in any way after starting the radar.

This is the crash report folder I got from a crash generated this morning: 1 hour and 10 minutes of normal functionning with ARPA ploted from 10m to 100m (I'm not at sea right now so I just generate ARPA in a room):
[crash_report.zip](https://github.com/opencpn-radar-pi/radar_pi/files/13203711/crash_report.zip)

This is the opencpn.ini file used:
[opencpn_ini.txt](https://github.com/opencpn-radar-pi/radar_pi/files/13203741/opencpn_ini.txt)

And this is the logfile generated during this test (from 8h53 to 10h04):
[opencpn_log.log](https://github.com/opencpn-radar-pi/radar_pi/files/13203752/opencpn_log.log)

I hope that I provided enough information. I have no idea about the cause of this issue. I've reinstalled OpenCPN recently and did not change much in its initial configuration appart from intalling the plugins Radar_Pi and NMEA_Converter_Pi.

Thanks for reading me, and have a nice day !

douwefokkema, 2023-10-30:
I just had the radar (Halo 20+) running for 4 hours on Windows, no crash. I have ARPA on but no NMEA_Converter_Pi. I will continue for some hours more. Same versions of software as you have. Question: Does it also crash with ARPA off? Does it crash without NMEA_Converter_Pi?

ERousseletChorus, 2023-10-31:
I've had the radar running for 2,5 hour this morning with ARPA OFF and NMEA_Converter_Pi enabled with no crash. I just started it with ARPA ON and NMEA_Converter_Pi disabled. Since I don't have a compass, I'm missing the heading parameter, so the radar is not able to detect ARPA in this configuration even if they are set ON.

I'll edit this comment when I have a result (2h30 with no crash or a crash after XXhXX).

Hakansv, 2023-10-31:
One thought:
Do you happens to have the PPI window open receiving AIS targets? If so you may try to switch off the PPI window.
There's a detected memory leak in O 5.8.4 for that case. (Now corrected in Github master)

ERousseletChorus, 2023-10-31:
> One thought: Do you happens to have the PPI window open receiving AIS targets? If so you may try to switch off the PPI window. There's a detected memory leak in O 5.8.4 for that case. (Now corrected in Github master)

Yes, I have the parameter AIS/ARPA on PPI ON. I start a test with yesterday's configuration and this parameter OFF immediatly !

ERousseletChorus, 2023-10-31:
> Yes, I have the parameter AIS/ARPA on PPI ON. I start a test with yesterday's configuration and this parameter OFF immediatly !

The radar has been running for 2,5 hours without any crash right now. So I think this was the issue. Sorry if the AIS/ARPA on PPI issue was already reported somewhere and I missed it.

Huge thanks to you two for helping me solving this issue, and have a great day !

closed 2023-10-31 by ERousseletChorus

Hakansv, 2023-10-31:
@ERousseletChorus 
You're welcome and sorry for the O bug creating this. You've to await a new OCPN release. (Technically it's possible to use a unstable alpha version from the ongoing developing work. But maybe not suitable for navigation?)