opencpn-radar-pi/radar_pi #252: Help connecting Raymarine Quantum 2 Q24C to openCPN

MaxwellJung, 2023-11-14:
I am having trouble getting openCPN to recognize my Raymarine Quantum 2 Q24C radar.

I have the Quantum radar connected directly to RPi via ethernet, but openCPN radar setup window just says "initializing".

I also tried plugging the radar to a router via ethernet then the RPi to the router via ethernet. The router seems to assign an ip address to the radar, but openCPN still can't seem to find the radar in the network.

Should I try running a DHCP server on the RPi itself?
Any help would be appreciated thanks.

jpcavanagh, 2023-11-14:
Are you trying to use the radar 'stand-alone' i.e. without an AXIOM in the
network?     I use the Raymarine Quantum II without problems, but the AXIOM
has to be turned on first, then the radar.  I understand from Raymarine
that when the radar is turned on (even from sleep mode) it spends 10
minutes looking for an Axiom, and if it fails to find one it shutdowns
down.  This is why I had to have a special power switch installed to turn
off/on the radar scanner.  If it shuts down, only a power cycle recovers it.

Once the Axiom is on, the radar comes up, then the PC can access the
radar.  If you then turn off the Axiom, the PC loses contact with the radar
after about 10 minutes.

John

dhruvna, 2023-11-14:
> Are you trying to use the radar 'stand-alone' i.e. without an AXIOM in the network? I use the Raymarine Quantum II without problems, but the AXIOM has to be turned on first, then the radar. I [...]

Is it possible for us to connect the two without using an AXIOM? Our goal right now is to connect the radar directly to the rpi.

MaxwellJung, 2023-11-16:
Got it working!

Turns out the setup window disappearing meant that the radar successfully initialized.

I just had to right click on the map to bring up the radar control window and looks like everything works.

Final setup:
connect radar to router
connect RPi to router
follow radarPi instructions

closed 2023-11-16 by MaxwellJung