opencpn-radar-pi/radar_pi #255: Garmin xHD not happy being only radar ...

rszemeti, 2024-01-17:
I am running OpenCPN 5.8.4, radar 5.5.0 ... with a Garmin GMR 18HD+ (note the plus ...) scanner.

If I select "Garmin xHD" as  type, doesnt work, shows the "disconnected" image as icon in side mene no options appear when clicking radar icon

If I  select any other radar type as a second radar (even just emulator, or a type I don't have) then it works fine.

rszemeti, 2024-01-18:
So a bit more fiddling around and investigation ... I cannot get it to display with a single radar type selected, no big deal, I just select two types and then close the PPI display of the one I dont want, leaving the Garmin xHD on screen ... sorted.

If I then stop/start OpenCPN everythig is fine. I can toggle the PPI display with the radar icon in the side menu, no problem.  I can close OpenCPN with the PPI shown or not shown and restart it in the same start, toggle it with the menu icon no problem.

Iff clear the PPI off the screen with the "close" icon top right of the PPI, I'm screwed. No amount of restarting OpenCPN or clicking on the menu item will restore either PPI display.

I *think* this is related to the Show=[1|0] and Radar1WindowShow=[1|0] settings ... 

I can sort of see the logic in that if you have removed a PPI you dont want ti popping back up again when you toggle the icon, but it does end up in locked out state where no PPI can be displayed and ther seems to be no recovery from this?

I may be mixing two issues here ...

rszemeti, 2024-01-18:
It seems I was confusing matters with the "two PPIs" thing .. once the PPI is closed with the [x] ther seems to be no way to get it back up again, deselecting/selecting a second radar does seem to manage to overwrite the settings, getting it back on screen.  I can paste a working/non-working config if needed.

To reproduce:

click icon to display PP on screen
dimiss PPI with the [x] top right of PPI display.
clicking radar icon does not recover PPI, no way to get back to radar menu to display the PPI again, effectively leaving it in an unworkable state.

rszemeti, 2024-01-18:
I think I understand the issue now ... the "icon" toggles between show radars (any individual PPIs or overlays marked as shown) and globally hides/shows them ...  but this makes no sense in a situation where ALL the idividual PPIs and overlays are hidden. 

For a user who has hidden the PPI the "wrong way" by clicking the [x] rather than the icon, it is only possible to get it back if you can right click, which I am finding rather tricky on a touchscreen.  I appreciate the idea of the [x] was to allow multi-scanner users to remove individual PPI's 

Would it not make more sense, in a situation where ALL PPI displays and overlays are hidden to show *something* when the icon is clicked  ...

Hakansv, 2024-01-18:
Have you tried the canvas menu? Right click the chart window. There are radar items as well.

mgrouch, 2024-01-29:
I see same issue here. If I close radar screen using top right [x] I can’t bring it back

miluk-nl, 2024-01-29:
Right mouse click on the chart, then click "Show radar" or "Radar Control..." followed by "Window > Show PPI".

rszemeti, 2024-01-29:
I have a touchscreen, right click is kinda difficult.

To my mind, if you turned off the PPI and you turned off the overlay .. now
the icon in the menu does nothing, that seems, from a UI perspective,
wrong. I would expect clicking the icon to do *something*  maybe if both
PPI and overlay are Off, when you toggle the menu item to "on" it should
enable them both.

miluk-nl, 2024-01-31:
> I have a touchscreen, right click is kinda difficult.

Long hold on the screen should give a square grey outline under your finger, then release. This should bring up the context menu.

douwefokkema, 2024-01-31:
The solution I proposed does not require a right click, it only makes use of the icon in the menu bar.

rszemeti, 2024-01-31:
> A possible solution could be to adapt the function of the radar icon on
> the menubar. Currently this button is mainly a hide / show radar button.
> And the "show" click will get back the same what you hided, if there were
> no radar items shown, you get nothing back. It is simple to make the button
> always show the radar controls. So if there are no radar items visible,
> click the icon, hide the nothing. Click again, and the radar controls for
> all radars are show. This would also help users that don't know about the
> canvas menu, and ask questions of the forum how to show the radar. But my
> suggested solution will irritate users who use the icon to hide / show the
> radar and get back exactly what they have hidden. Now they will also get
> the radar controls. But these users could switch to the canvas menu hide /
> show radar.
> I have pushed a test version of this change in the branch "menubar". This
> change could possibly be implemented in a next release.

A solution that would solve this issue and not irritate the hide show/users
is simple, there are three setting to take into account:

  1) the global visibility setting

  2) the individual PPI visibility

  3) the individual overlay visibility

Clicking the icon toggles the global visibility.  This is the current
situation

Proposed solution:
if either PPI or overlay are individually marked as shown, then clicking
the icon toggles global visibility, same as current situation

if both PPI and Overlay are both marked individually as hidden:
     Current situation:  Clicking icon toggles global visibility, but no
effect because both items are individually marked as hidden.
     Proposed solution:  Clicking icon sets both PPI and overlay to
visible, toggles global visibility

keesverruijt, 2024-08-14:
Fixed in v5.5.4 commit d1b32e42c769ebf55cccfadf97e4e1d26330e7c0 

Just press the Radar icon in the icon list, when radars are hidden it will show the controls.

closed 2024-10-02 by keesverruijt