opencpn-radar-pi/radar_pi #256: Garmin xHDnot shutting down on OpenCPN close

rszemeti, 2024-01-17:
I am running OpenCPN 5.8.4, radar 5.5.0 ... with a Garmin GMR 18HD+ selected as xHD type.

If the radar is on and active, when I close OpenCPN it does not send a stop command and the radar keeps on spinning ...

rszemeti, 2024-01-17:
I don't have any other stations looking at data,  I suspect 99% of installations are a single station.

Would it not be possible to have a "stop radar on close" checkbox .. or better still use a GPIO to close the switch when the radar plugin is active?

Adding manual switches to do things is fine, but when we have a computer that can do it for us (and not forget to do it ... which is what happens when you give the job to a human ... )

Hakansv, 2024-01-17:
It seems you're on a RPi? Why not investigate Tony's (Antipole) plugin "JavaScript " to serve this?

rszemeti, 2024-01-17:
Let me try leaving it running and see if it stops itself after some time, maybe it is a non-issue.

rszemeti, 2024-01-17:
Well, it's still whirring away an hour later, so let's put that down as a "nope" then.

I've had a look at the Javascript plugin, but struggling to work out how to get it a) trigger on application close, b) keep the application alive until it has finsihed doing whatever it needs to do and c) call the appropriate method in the radar plugin!