opencpn-radar-pi/radar_pi #258: Unable to build in Fedora 36

opaezosuna, 2024-02-13:
Hello,

It appears that some files are missing from my build tree. After downloading the plugin repo and executing cmake I get:

```
-- *** Added plugin: /home/opo/git/OpenCPN/plugins/grib_pi
CMake Error at plugins/radar_pi/CMakeLists.txt:15 (include):
  include could not find requested file:

    /home/opo/git/OpenCPN/cmake/CmakeSetup.cmake

-- Cmake version: 3.26.3.
CMake Error at plugins/radar_pi/CMakeLists.txt:21 (include):
  include could not find requested file:

    PluginOptions

-- Configuring
CMake Error at plugins/radar_pi/CMakeLists.txt:35 (include):
  include could not find requested file:

    PluginCompiler

CMake Error at plugins/radar_pi/CMakeLists.txt:40 (add_subdirectory):
  add_subdirectory given source "opencpn-libs/api-16" which is not an
  existing directory.

CMake Error at plugins/radar_pi/CMakeLists.txt:54 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.

CMake Error at plugins/radar_pi/CMakeLists.txt:65 (include):
  include could not find requested file:

    ConfigureManifest

CMake Error at plugins/radar_pi/CMakeLists.txt:66 (configure_manifest):
  Unknown CMake command "configure_manifest".

-- Configuring incomplete, errors occurred!

```
Please provide any pointers on what I am missing here, I did not have any issues building other plugins (e.g. rtlsdr_pi) or opencpn itself. This happens from trying to build from the master and from 5.5.0 version branches.

Thanks!

nohal, 2024-02-13:
Seeing the mention of `grib_pi` and the paths It looks like you are trying to build the plugin inside the OpenCPN sources. That is not how it is supposed to be done, this is a completely separate project. All the third party plugins are.

opaezosuna, 2024-02-13:
Thank you @nohal ,and yes I was trying to build it as part of OpenCPN because the one that I independently built was being rejected by OpenCPN during the import process as a non-compatible plugin. I will read the documentation on how to import 3rd party plugins.

nohal, 2024-02-13:
You may have to play with the plugin metadata and the system identifier as Fedora is not in the list of suported distributions. Locally you may just copy the `.so` and data files directory and are done. Or use flatpak and you won't have to build anything...

opaezosuna, 2024-02-13:
Thanks, it seems to be running now after manually copying the files. I'll be getting a donated Raymarine RD418D very soon which I hope I can make it work on my boat (running OpenCPN on a pi4). Before I take it to the boat I'll try to getting working here and bench test at home first. From the radar list, I do not see any options listed as Raymarine RXXXD, do you know which is the one compatible with it E120 or Quantum?

closed 2024-03-10 by keesverruijt