opencpn-radar-pi/radar_pi #259: garmin xhd3

tkellen, 2024-03-10:
Hello! Can you please confirm if this radar is supported by your plugin?
https://www.garmin.com/en-AU/p/951697

If not, can you please advise what the major blockers are to making it so? I would be glad to participate in the development needed to add support if there is none.

tkellen, 2024-03-10:
In a similar situation here. Just purchased a xHD3 and can't get it to power on shorting pin 5 to the drain wire/shield. cc @trudK45 in the hopes you can help. I'm very motivated to assist in any dev needed (it any is needed) once I get the damn thing to turn on.

keesverruijt, 2024-03-10:
Hmm, it looks as if Garmin changed something. Your best bet is to get (temporary access to) a Garmin MFD plus some computer running tcpdump/wireshark, and analyse how Garmin does the startup. I've always thought the physical shorting of pin 5 was crude, they probably changed to a WOL (Wake On Lan) packet or something else like that.

keesverruijt, 2024-03-10:
Even if you do it is likely that the xHD3 will not work fully. Until you fix it yourself, the answer is "no".

tkellen, 2024-03-10:
No "if", I will be pursuing this. I'll happily use this thread to post my progress if it is welcome. Would you accept contributions for this device?

Current state of affairs:

Connecting the unit to 12v power with an ammeter clamped over the positive lead shows a momentary spike of 0.5amps that drops to 0.1 shortly after.

tshark shows udp packets intermittently on an ethernet interface using 172.16.0.0/255.255.240.0. This works with both crossover and standard ethernet pinouts.

As mentioned previously, shorting pin 5 to the drain wire/cable shield does nothing. Will try a wol packet in a few hours. Heading to bed here, I am GMT+10 and it is too late/early to continue.

keesverruijt, 2024-03-10:
Sure, all contributions are super welcome!

tharapita, 2024-05-03:
> In a similar situation to #212 here. Just purchased a xHD3 and can't get it to power on shorting pin 5 to the drain wire/shield. cc @trudK45 in the hopes you can help. 

@tkellen  Did you get xHD3 to work? I'm not having luck finding old xHD suppliers (just got a refund from the second seller I tried because they could not source it despite claiming to have it in stock).

Pretty likely that I'll order a HD3 or xHD3 direct from Garmin as the next step. I also have a gpsmap plotter from Garmin to debug, but will need to move it off the boat in a few months.

tkellen, 2024-05-03:
just finished my refit and finally left sailing north along the coast of Australia to Indonesia this year. I haven't gotten to it yet but it is sitting on my mast unused and eventually I will. I've been a software developer for 25 years, nothing about getting this thing to work sounds hard once I've figured out how to turn it on 😂

tkellen, 2024-05-03:
I will absolutely provide updates and a PR to this repo eventually, unless someone gets to it before me, which seems unlikely.

tharapita, 2024-05-03:
> just finished my refit and finally left sailing north along the coast of Australia to Indonesia this year. I haven't gotten to it yet but it is sitting on my mast unused and eventually I will. I've [...]

Hehe :)

But great, thanks.

Hakansv, 2026-01-20:
Adding another radar type I've not been part of but you need e.g. Wireshark to see each command from the MFD and corresponding answers from the scanner. Something is written [here ](https://github.com/opencpn-radar-pi/radar_pi/wiki/Adding-new-radar-types)and example captures for other radar types are [here](https://github.com/opencpn-radar-pi/radar_pi/tree/master/example)