opencpn-radar-pi/radar_pi #260: Run standalone possible?

sailing12388, 2024-04-22:
Is it possible to run this plugin as a standalone application without OpenCPN running?

Hakansv, 2024-04-22:
Quick answer: No.
Pls close this. Issues is about bugs and similar about the program.
Thanks.

keesverruijt, 2024-04-22:
No, it is not. 

There is talk of creating a completely new implementation, which would be more independent from OpenCPN. 

What is your use case?

sailing12388, 2024-04-22:
> No, it is not.
> 
> There is talk of creating a completely new implementation, which would be more independent from OpenCPN.
> 
> What is your use case?

Our use case is to control our B&G radar on our nav desk PC. This PC runs navigation software that is not OPCPN. We do run it at the moment but minimized so we can view/control the radar.

sailing12388, 2024-06-30:
> No, it is not.
> 
> There is talk of creating a completely new implementation, which would be more independent from OpenCPN.
> 
> What is your use case?

Also, if anyone is looking into making this stand alone, we would be open to help fund that endeavor.

keesverruijt, 2024-10-02:
Hi @sailing12388 the new implementation is now in active development, see https://github.com/keesverruijt/mayara
Feel free to join our Discord server: https://discord.gg/kC6h6JVxxC

closed 2024-10-02 by keesverruijt