opencpn-radar-pi/radar_pi #261: Raymarine RD418D

opaezosuna, 2024-04-30:
Hello,
Currently I am bench testing connecting both the raspberry pi and the RD418D radar to an ethernet hub. The plugin says
interface 169.254.37.140 error: cannot bind UDP socket to port 1.

Before opening the radar to verify/redo the ethernet cable, am I missing any step here? The radar is powered and spinning.

Thanks!

keesverruijt, 2024-05-01:
The plugin tries to connect to the radar on all ethernet interfaces. Is the interface with IP address 169.254.37.140 the interface that is connected to the radar (via switch(es))?

keesverruijt, 2024-05-01:
Also the port number may be an issue, what happens if you run this under `root`?

jpcavanagh, 2024-05-01:
An IP 169.254.x.x is in the automatic private IP address range, ie it was self allocated.  I suspect the Raymarine radar is waiting to talk to a MFD to be allocated an IP address.  Many Raymarine ethernet radars need to acquire an IP address from a MFD within 10 mins, otherwise they go into 'sleep mode' which often requires a power recycle to recover from.

Either way I suspect this is more of a hardware issue than a software one.

opaezosuna, 2024-05-01:
Thank you all for your responses. I will try running as root to verify if it is a permission issue. I've tried with different routers (with auto IP and with self IP) obtaining similar results:
interface 192.168.1.190 error: cannot bind UDP socket to port 1 (this is on day to day local network)

Here are some relevant corresponding log entries are:
```
20:04:45.183 MESSAGE pluginmanager.cpp:1168 Checking plugin candidate: /usr/local/lib64/opencpn/libradar_pi.so
20:04:45.183 MESSAGE pluginmanager.cpp:2179 Plugin is compatible by elf library scan: true
20:04:45.183 MESSAGE pluginmanager.cpp:2318 PlugInManager: Loading PlugIn: /usr/local/lib64/opencpn/libradar_pi.so
20:04:45.186    INFO pluginmanager.cpp:2494 PlugInManager:  /usr/local/lib64/opencpn/libradar_pi.so
20:04:45.186    INFO pluginmanager.cpp:2495         Plugin common name: Radar
20:04:45.186    INFO pluginmanager.cpp:2497         API Version detected: 116
20:04:45.186    INFO pluginmanager.cpp:2498         PlugIn Version detected: 5.5
20:04:45.186 MESSAGE OCPNPlatform.cpp:1672 Using plugin data path: /home/opo/.local/share/opencpn/plugins;/home/opo/.local/share/flatpak/exports/share/opencpn/plugins;/var/lib/flatpak/exports/share/opencpn/plugins;/usr/local/share/opencpn/plugins;/usr/share/opencpn/plugins;/usr/local/lib64/opencpn
20:04:45.186 MESSAGE pluginmanager.cpp:572 PlugInManager: Using data dirs from: /home/opo/.local/share/opencpn/plugins;/home/opo/.local/share/flatpak/exports/share/opencpn/plugins;/var/lib/flatpak/exports/share/opencpn/plugins;/usr/local/share/opencpn/plugins;/usr/share/opencpn/plugins;/usr/local/lib64/opencpn
20:04:45.186 MESSAGE pluginmanager.cpp:584 PlugInManager: using data dir: /usr/local/share/opencpn/plugins/radar_pi
20:04:45.192 MESSAGE radar_pi.cpp:280 5.5.2-alpha+2402130310.576545a
20:04:45.192    INFO radar_pi.cpp:1465 Detected OpenCPN Version 5.6.1
20:04:45.192    INFO radar_pi.cpp:1590 Config Loaded RadarCount=1
20:04:45.192    INFO radar_pi.cpp:1696 Config Save RadarCount=1
20:04:45.192 MESSAGE radar_pi.cpp:302 Configuration file values initialised
20:04:45.192 MESSAGE radar_pi.cpp:303 Log verbosity = 0. To modify, set VerboseLog to sum of:
20:04:45.192 MESSAGE radar_pi.cpp:304 VERBOSE  = 1
20:04:45.192 MESSAGE radar_pi.cpp:305 DIALOG   = 2
20:04:45.192 MESSAGE radar_pi.cpp:306 TRANSMIT = 4
20:04:45.192 MESSAGE radar_pi.cpp:307 RECEIVE  = 8
20:04:45.192 MESSAGE radar_pi.cpp:308 GUARD    = 16
20:04:45.192 MESSAGE radar_pi.cpp:309 ARPA     = 32
20:04:45.192 MESSAGE radar_pi.cpp:310 REPORTS  = 64
20:04:45.210 MESSAGE glChartCanvas.cpp:732 BuildFBO 3
20:04:45.211 MESSAGE glChartCanvas.cpp:1217 OpenGL Framebuffer OK, size = 1918
20:04:45.214 MESSAGE RaymarineReceive.h:74 Raymarine E120 receive thread created, prio= 70
20:04:45.214 MESSAGE RaymarineReceive.h:95 radar addresses from ini file loaded
20:04:45.214 MESSAGE RaymarineReceive.h:97  Raymarine E120 using addresses:  /236.6.7.9:1/236.6.7.9:1/
20:04:45.214 MESSAGE radar_pi.cpp:405 radar_pi Raymarine locator started
20:04:45.214 MESSAGE RaymarineLocate.cpp:137 RaymarineLocate thread starting
20:04:45.215   ERROR RaymarineLocate.cpp:96 Attempting to start receive socket on  224.0.0.1:5800
20:04:45.215   ERROR RaymarineLocate.cpp:103 Attempting to start receive socket on  232.1.1.1:5800
20:04:45.215   ERROR RaymarineLocate.cpp:96 Attempting to start receive socket on  224.0.0.1:5800
20:04:45.215   ERROR RaymarineLocate.cpp:103 Attempting to start receive socket on  232.1.1.1:5800
20:04:46.215   ERROR RaymarineReceive.cpp:162 Raymarine E120 Unable to listen to socket:  Cannot bind UDP socket to port 1
20:04:46.215 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - -1
20:04:46.465 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 0
20:04:46.465   ERROR RaymarineReceive.cpp:162 Raymarine E120 Unable to listen to socket:  Cannot bind UDP socket to port 1
20:04:46.465 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - -1
20:04:46.718 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 1
20:04:46.718   ERROR RaymarineReceive.cpp:162 Raymarine E120 Unable to listen to socket:  Cannot bind UDP socket to port 1
20:04:46.718 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - -1
20:04:46.998 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 2
20:04:46.998   ERROR RaymarineReceive.cpp:162 Raymarine E120 Unable to listen to socket:  Cannot bind UDP socket to port 1
20:04:46.998 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - -1
20:04:47.216 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - -1
20:04:47.467 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 4
20:04:47.467   ERROR RaymarineReceive.cpp:162 Raymarine E120 Unable to listen to socket:  Cannot bind UDP socket to port 1
20:04:47.467 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - -1
20:04:47.717 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 5
[... cut ...]
20:04:48.976 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - -1
20:04:49.226 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 2
20:04:49.227   ERROR RaymarineReceive.cpp:162 Raymarine E120 Unable to listen to socket:  Cannot bind UDP socket to port 1
20:04:49.227 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - -1
20:04:49.477 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 3
```

opaezosuna, 2024-05-01:
When running as root, I get these extra lines:
```
10:20:56.012 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - -1
10:20:56.012   ERROR RaymarineLocate.cpp:96 Attempting to start receive socket on  224.0.0.1:5800
10:20:56.012   ERROR RaymarineLocate.cpp:103 Attempting to start receive socket on  232.1.1.1:5800
10:20:56.012   ERROR RaymarineLocate.cpp:96 Attempting to start receive socket on  224.0.0.1:5800
10:20:56.012   ERROR RaymarineLocate.cpp:103 Attempting to start receive socket on  232.1.1.1:5800
10:20:56.012 MESSAGE RadarInfo.cpp:156 Raymarine E120 receive thread stopped in 0 ms
10:20:56.014    INFO radar_pi.cpp:1696 Config Save RadarCount=1
10:20:56.016 MESSAGE radar_pi.cpp:474 radar_pi Context menus removed
10:20:56.028 MESSAGE chart1.cpp:3826 opencpn::MyFrame exiting cleanly.
10:20:56.041 MESSAGE chartdb.cpp:282 Chart cache PlugIn purge
10:20:56.043 MESSAGE chart1.cpp:4004 gFrame destroyed.
10:20:56.055 MESSAGE chart1.cpp:2627 opencpn::MyApp starting exit.
10:20:56.055 MESSAGE chart1.cpp:2666 LOGBOOK:  2024-05-01 14:20:56 UTC OFF: Lat   33.35800 Lon  -79.28200
10:20:56.067 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: /usr/local/share/opencpn/tcdata/harmonics-dwf-20210110-free.tcd
10:20:56.072 MESSAGE TCDataSource.cpp:41 UnLoading Tide/Current data source: /usr/local/share/opencpn/tcdata/HARMONICS_NO_US.IDX
10:20:56.072 MESSAGE chart1.cpp:2690 opencpn::MyApp exiting cleanly...

```
I'll keep looking.

opaezosuna, 2024-05-01:
How long does it usually take the first time to find the radar? Also, should the radar's IP be visible in the network table?

opaezosuna, 2024-05-01:
Log info when running as root, sorry for the long post:
```
11:07:57.283 MESSAGE pluginmanager.cpp:1168 Checking plugin candidate: /usr/local/lib64/opencpn/libradar_pi.so
11:07:57.283 MESSAGE pluginmanager.cpp:2179 Plugin is compatible by elf library scan: true
11:07:57.283 MESSAGE pluginmanager.cpp:2318 PlugInManager: Loading PlugIn: /usr/local/lib64/opencpn/libradar_pi.so
11:07:57.285    INFO pluginmanager.cpp:2494 PlugInManager:  /usr/local/lib64/opencpn/libradar_pi.so
11:07:57.285    INFO pluginmanager.cpp:2495         Plugin common name: Radar
11:07:57.285    INFO pluginmanager.cpp:2497         API Version detected: 116
11:07:57.285    INFO pluginmanager.cpp:2498         PlugIn Version detected: 5.5
11:07:57.292 MESSAGE OCPNPlatform.cpp:1672 Using plugin data path: /root/.local/share/opencpn/plugins;/usr/local/share/opencpn/plugins;/usr/share/opencpn/plugins;/usr/local/lib64/opencpn
11:07:57.292 MESSAGE pluginmanager.cpp:572 PlugInManager: Using data dirs from: /root/.local/share/opencpn/plugins;/usr/local/share/opencpn/plugins;/usr/share/opencpn/plugins;/usr/local/lib64/opencpn
11:07:57.292 MESSAGE pluginmanager.cpp:584 PlugInManager: using data dir: /usr/local/share/opencpn/plugins/radar_pi
11:07:57.298 MESSAGE radar_pi.cpp:280 5.5.2-alpha+2402130310.576545a
11:07:57.298    INFO radar_pi.cpp:1465 Detected OpenCPN Version 5.6.1
11:07:57.298    INFO radar_pi.cpp:1590 Config Loaded RadarCount=1
11:07:57.299    INFO radar_pi.cpp:1696 Config Save RadarCount=1
11:07:57.299 MESSAGE radar_pi.cpp:302 Configuration file values initialised
11:07:57.299 MESSAGE radar_pi.cpp:303 Log verbosity = 0. To modify, set VerboseLog to sum of:
11:07:57.299 MESSAGE radar_pi.cpp:304 VERBOSE  = 1
11:07:57.299 MESSAGE radar_pi.cpp:305 DIALOG   = 2
11:07:57.299 MESSAGE radar_pi.cpp:306 TRANSMIT = 4
11:07:57.299 MESSAGE radar_pi.cpp:307 RECEIVE  = 8
11:07:57.299 MESSAGE radar_pi.cpp:308 GUARD    = 16
11:07:57.299 MESSAGE radar_pi.cpp:309 ARPA     = 32
11:07:57.299 MESSAGE radar_pi.cpp:310 REPORTS  = 64
11:07:57.317 MESSAGE glChartCanvas.cpp:732 BuildFBO 3
11:07:57.317 MESSAGE glChartCanvas.cpp:1217 OpenGL Framebuffer OK, size = 1342
11:07:57.320 MESSAGE RaymarineReceive.h:74 Raymarine E120 receive thread created, prio= 70
11:07:57.320 MESSAGE RaymarineReceive.h:95 radar addresses from ini file loaded
11:07:57.320 MESSAGE RaymarineReceive.h:97  Raymarine E120 using addresses:  /236.6.7.9:1/236.6.7.9:1/
11:07:57.320 MESSAGE radar_pi.cpp:405 radar_pi Raymarine locator started
11:07:57.320 MESSAGE RaymarineLocate.cpp:137 RaymarineLocate thread starting
11:07:57.344 MESSAGE OCPNPlatform.cpp:1106 Loaded translation catalog for:  opencpn-radar_pi
11:07:57.350   ERROR RaymarineLocate.cpp:96 Attempting to start receive socket on  224.0.0.1:5800
11:07:57.350   ERROR RaymarineLocate.cpp:103 Attempting to start receive socket on  232.1.1.1:5800
11:07:57.350   ERROR RaymarineLocate.cpp:96 Attempting to start receive socket on  224.0.0.1:5800
11:07:57.350   ERROR RaymarineLocate.cpp:103 Attempting to start receive socket on  232.1.1.1:5800
11:07:58.333 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - 43
11:07:58.467 MESSAGE chart1.cpp:7243    ***GPS Watchdog timeout at Lat:33.358   Lon: -79.282
11:07:58.572 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 0
11:07:58.572   ERROR RME120Control.cpp:95 Raymarine E120 Unable to transmit command to unknown radar
11:07:58.572   ERROR RME120Control.cpp:95 Raymarine E120 Unable to transmit command to unknown radar
11:07:58.821 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 1
11:07:59.072 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 2
11:07:59.072   ERROR RME120Control.cpp:95 Raymarine E120 Unable to transmit command to unknown radar
11:07:59.322 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 3
11:07:59.572 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 4
11:07:59.823 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 5
11:08:00.073 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 6
11:08:00.073   ERROR RME120Control.cpp:95 Raymarine E120 Unable to transmit command to unknown radar
11:08:00.324 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 7
11:08:00.583 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 8
11:08:00.583 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - 43
11:08:00.827 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 0
11:08:01.077 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 1
11:08:01.077   ERROR RME120Control.cpp:95 Raymarine E120 Unable to transmit command to unknown radar
11:08:01.328 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 2
11:08:01.578 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 3
11:08:01.828 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 4
11:08:02.079 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 5
[... cut ...]
11:08:56.757 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 6
11:08:57.007 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 7
11:08:57.007   ERROR RME120Control.cpp:95 Raymarine E120 Unable to transmit command to unknown radar
11:08:57.257 MESSAGE RaymarineReceive.cpp:274 Raymarine E120 RaymarineReceive receive timeout 8
11:08:57.266 MESSAGE RaymarineReceive.cpp:112 Raymarine E120 Creating multicast socket for radar at IP 0.0.0.0 port 0 [ /236.6.7.9:1/236.6.7.9:1/] - 43

```

opaezosuna, 2024-05-01:
> An IP 169.254.x.x is in the automatic private IP address range, ie it was self allocated. I suspect the Raymarine radar is waiting to talk to a MFD to be allocated an IP address. Many Raymarine [...]

@jpcavanagh Thank you for your comment. I do not see the radar's IP in the table after connecting its network cable to the router. I will try with other network cables once I open the radome to see if it is a cable related issue. Do you know if it is a standard ethernet cable that is used in this case? From Raymarine's procedures it is suggested that there is an standard RJ45 plug inside to which the bundled ethernet cable is connected.

Thanks again!

thwd, 2024-05-21:
Hi, I have the exact same radar and setup. Same exact results, running as root as well.

If you want me to do tests, I'm available.

thwd, 2024-05-21:
Yes, I have a fully functional 12VDC battery array operating at ~13.8V and capable of something like 150A between them. The (original Raymarine installation) cables I found connected to the RD418D are 4mm diameter. I am in the middle of a refit so I have connected them straight to battery. I have no concerns about power.

I am using a Raymarine crossover adapter to connect the RJ45 from the radome to my Rapsberry Pi 4 running OpenCPN with Radar Pi.

opaezosuna, 2024-05-21:
If you could also bring a switch or hub over that will also help in testing.

I'm no expert, but what I would try is to open a terminal, and run the commands
ifconfig
arp -a
also if you bring a switch/hub try logging into it after connecting the radar to it to see if the radar is listed as a connection, also connect the pi to it and run above commands. Try with both the raymarine adapter and without it.
Perhaps someone else in here would have better ideas to test.
Also share the opencpn logs when the pi is connected to either the radar or the switch.

opaezosuna, 2024-05-21:
Also, if you have not done it yet, try testing the voltage at the radome end of the cable bundle (two central pins) maybe the voltage there is too low?

thwd, 2024-05-27:
After trying a lot of different things, I did a package capture with Wireshark (attached below, also as text).

I tested the radar's power -- it's fine throughout.

I tried connecting the radar straight-through to a commerical wifi router. It doesn't pick it up. I tried inserting the crossover adapter in between, also nothing.

I connected the radar with the crossover adapter to my Pi running openplotter and took the following capture:

[radar.pcapng.zip](https://github.com/opencpn-radar-pi/radar_pi/files/15459088/radar.pcapng.zip)

```
No.	Time	Source	Destination	Protocol	Length	Info
1	0.000000000	0.0.0.0	255.255.255.255	DHCP	342	DHCP Discover - Transaction ID 0xc7575d41
2	2.982442578	169.254.41.117	239.2.1.1	UDP	439	2053 → 2053 Len=397
3	8.576010905	::	ff02::16	ICMPv6	110	Multicast Listener Report Message v2
4	8.795977342	::	ff02::1:ff06:bdde	ICMPv6	86	Neighbor Solicitation for fe80::39a8:a0bb:4506:bdde
5	8.811947578	0.0.0.0	255.255.255.255	DHCP	342	DHCP Discover - Transaction ID 0x1eae6d5e
6	9.447949923	::	ff02::16	ICMPv6	110	Multicast Listener Report Message v2
7	9.869322900	fe80::39a8:a0bb:4506:bdde	ff02::16	ICMPv6	110	Multicast Listener Report Message v2
8	9.870102617	fe80::39a8:a0bb:4506:bdde	ff02::2	ICMPv6	70	Router Solicitation from d8:3a:dd:99:42:67
9	9.896522697	fe80::39a8:a0bb:4506:bdde	ff02::16	ICMPv6	110	Multicast Listener Report Message v2
10	10.555247163	fe80::39a8:a0bb:4506:bdde	ff02::fb	MDNS	1441	Standard query response 0x0000 TXT, cache flush AAAA, cache flush fe80::39a8:a0bb:4506:bdde PTR, cache flush openplotter.local SRV, cache flush 0 0 3000 openplotter.local TXT, cache flush SRV, cache flush 0 0 10110 openplotter.local TXT, cache flush SRV, cache flush 0 0 3000 openplotter.local TXT, cache flush SRV, cache flush 0 0 8375 openplotter.local TXT, cache flush SRV, cache flush 0 0 3000 openplotter.local
11	11.667837788	fe80::39a8:a0bb:4506:bdde	ff02::fb	MDNS	107	Standard query 0x0000 PTR _ipps._tcp.local, "QM" question PTR _ipp._tcp.local, "QM" question
12	11.986092662	fe80::39a8:a0bb:4506:bdde	ff02::fb	MDNS	1498	Standard query response 0x0000 PTR _signalk-ws._tcp.local PTR openplotter._http._tcp.local TXT, cache flush SRV, cache flush 0 0 3000 openplotter.local AAAA, cache flush fe80::39a8:a0bb:4506:bdde PTR _http._tcp.local PTR openplotter._nmea-0183._tcp.local TXT, cache flush SRV, cache flush 0 0 10110 openplotter.local PTR _nmea-0183._tcp.local PTR openplotter._signalk-http._tcp.local TXT, cache flush SRV, cache flush 0 0 3000 openplotter.local PTR _signalk-http._tcp.local PTR openplotter._signalk-tcp._tcp.local TXT, cache flush SRV, cache flush 0 0 8375 openplotter.local PTR _signalk-tcp._tcp.local PTR openplotter._signalk-ws._tcp.local TXT, cache flush
13	12.053887749	fe80::39a8:a0bb:4506:bdde	ff02::fb	MDNS	140	Standard query response 0x0000 SRV, cache flush 0 0 3000 openplotter.local
14	12.672322775	fe80::39a8:a0bb:4506:bdde	ff02::fb	MDNS	1441	Standard query response 0x0000 TXT, cache flush AAAA, cache flush fe80::39a8:a0bb:4506:bdde PTR, cache flush openplotter.local SRV, cache flush 0 0 3000 openplotter.local TXT, cache flush SRV, cache flush 0 0 10110 openplotter.local TXT, cache flush SRV, cache flush 0 0 3000 openplotter.local TXT, cache flush SRV, cache flush 0 0 8375 openplotter.local TXT, cache flush SRV, cache flush 0 0 3000 openplotter.local
15	13.398959346	0.0.0.0	255.255.255.255	DHCP	342	DHCP Discover - Transaction ID 0x1eae6d5e
16	13.871556187	fe80::39a8:a0bb:4506:bdde	ff02::2	ICMPv6	70	Router Solicitation from d8:3a:dd:99:42:67
17	13.900736213	RaspberryPiT_99:42:67	Broadcast	ARP	42	Who has 169.254.41.117? (ARP Probe)
18	15.670126852	fe80::39a8:a0bb:4506:bdde	ff02::fb	MDNS	107	Standard query 0x0000 PTR _ipps._tcp.local, "QM" question PTR _ipp._tcp.local, "QM" question
19	15.844818362	RaspberryPiT_99:42:67	Broadcast	ARP	42	Who has 169.254.41.117? (ARP Probe)
20	17.577000691	RaspberryPiT_99:42:67	Broadcast	ARP	42	Who has 169.254.41.117? (ARP Probe)
21	17.872416631	fe80::39a8:a0bb:4506:bdde	ff02::2	ICMPv6	70	Router Solicitation from d8:3a:dd:99:42:67
22	19.579784778	RaspberryPiT_99:42:67	Broadcast	ARP	42	ARP Announcement for 169.254.41.117
23	19.588468947	169.254.41.117	224.0.0.22	IGMPv3	54	Membership Report / Join group 224.0.0.251 for any sources
24	19.638966399	169.254.41.117	224.0.0.251	MDNS	343	Standard query response 0x0000 PTR _signalk-ws._tcp.local PTR openplotter._http._tcp.local PTR _http._tcp.local PTR openplotter._nmea-0183._tcp.local PTR _nmea-0183._tcp.local PTR openplotter._signalk-http._tcp.local PTR _signalk-http._tcp.local PTR openplotter._signalk-tcp._tcp.local PTR _signalk-tcp._tcp.local PTR openplotter._signalk-ws._tcp.local
25	19.657585119	169.254.41.117	224.0.0.251	MDNS	1015	Standard query 0x0000 ANY e.d.d.b.6.0.5.4.b.b.0.a.8.a.9.3.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa, "QM" question ANY openplotter.local, "QM" question ANY 117.41.254.169.in-addr.arpa, "QM" question ANY openplotter._http._tcp.local, "QM" question ANY openplotter._nmea-0183._tcp.local, "QM" question ANY openplotter._signalk-http._tcp.local, "QM" question A 169.254.41.117 PTR openplotter.local SRV 0 0 3000 openplotter.local TXT SRV 0 0 10110 openplotter.local TXT SRV 0 0 3000 openplotter.local TXT AAAA fe80::39a8:a0bb:4506:bdde PTR openplotter.local
26	19.657715192	169.254.41.117	224.0.0.251	MDNS	583	Standard query 0x0000 ANY openplotter._signalk-ws._tcp.local, "QM" question ANY openplotter._signalk-tcp._tcp.local, "QM" question SRV 0 0 8375 openplotter.local TXT SRV 0 0 [... cut ...]
69	153.102798570	169.254.41.117	239.2.1.1	UDP	439	2053 → 2053 Len=397
70	163.104886911	169.254.41.117	239.2.1.1	UDP	439	2053 → 2053 Len=397
71	173.103720347	169.254.41.117	239.2.1.1	UDP	439	2053 → 2053 Len=397
72	183.104208235	169.254.41.117	239.2.1.1	UDP	439	2053 → 2053 Len=397
73	193.077811907	169.254.41.117	239.2.1.1	UDP	439	2053 → 2053 Len=397
74	199.575751776	0.0.0.0	255.255.255.255	DHCP	342	DHCP Discover - Transaction ID 0x1eae6d5e
75	203.075786773	169.254.41.117	239.2.1.1	UDP	439	2053 → 2053 Len=397
```

opaezosuna, 2024-05-28:
Thanks for sharing @thwd , I'll take a closer look at your output in the coming days. I'm thinking it might be necessary to connect an E120 or compatible chart plotter to capture the handshake process, perhaps a specific IP range or port is used. I do not have access to any of these hardware.

Maybe a developer might shed light on where in the code I should start looking at for this kind of effort (e.g searching for a wider range of addresses/ports), I have some C++ coding experience, not related to network issues though, and I am willing to dedicate some time for coding whatever is necessary to make some progress on this.

Is the networking/connection part of the plugin designed to act as an E120 (family) chart plotter?

opaezosuna, 2024-09-20:
Here is some capture of tcpdump after waiting for a long time with the radar off, then as soon as the radar was turned on I got this activity:

```
16:41:54.536900 IP 0.0.0.0.bootpc > 255.255.255.255.bootps: BOOTP/DHCP, Request from 3a:7f:58:51:00:fe (oui Unknown), length 282
16:41:54.541022 IP6 :: > ff02::16: HBH ICMP6, multicast listener report v2, 2 group record(s), length 48
16:41:54.959123 IP6 :: > ff02::1:ff53:d211: ICMP6, neighbor solicitation, who has fedora, length 32
16:41:55.079133 IP6 :: > ff02::16: HBH ICMP6, multicast listener report v2, 2 group record(s), length 48
16:41:55.975207 IP6 fedora > ff02::16: HBH ICMP6, multicast listener report v2, 2 group record(s), length 48
16:41:55.978115 IP6 fedora > ff02::16: HBH ICMP6, multicast listener report v2, 1 group record(s), length 28
16:41:56.079127 IP6 fedora > ff02::16: HBH ICMP6, multicast listener report v2, 3 group record(s), length 68
16:41:56.165981 IP6 fedora > ff02::2: ICMP6, router solicitation, length 8
16:41:56.271123 IP6 fedora > ff02::16: HBH ICMP6, multicast listener report v2, 1 group record(s), length 28
16:41:56.331775 IP6 fedora.mdns > ff02::fb.mdns: 0*- [0q] 2/0/0 (Cache flush) PTR fedora.local., (Cache flush) AAAA fe80::1fad:b176:1253:d211 (138)
16:41:56.537000 IP 0.0.0.0.bootpc > 255.255.255.255.bootps: BOOTP/DHCP, Request from 3a:7f:58:51:00:fe (oui Unknown), length 282
16:41:56.546893 ARP, Request who-has 10.10.10.110 tell 10.10.10.1, length 28
16:41:57.559333 ARP, Request who-has 10.10.10.110 tell 10.10.10.1, length 28
16:41:58.354548 IP6 fedora.mdns > ff02::fb.mdns: 0*- [0q] 2/0/0 (Cache flush) PTR fedora.local., (Cache flush) AAAA fe80::1fad:b176:1253:d211 (138)
16:41:58.584876 ARP, Request who-has 10.10.10.110 tell 10.10.10.1, length 28
16:41:58.668570 IP 0.0.0.0.bootpc > 255.255.255.255.bootps: BOOTP/DHCP, Request from 3a:7f:58:51:00:fe (oui Unknown), length 282
16:41:59.542910 IP 10.10.10.1 > 10.10.10.110: ICMP echo request, id 36823, seq 0, length 28
16:41:59.542975 IP 10.10.10.1.bootps > 10.10.10.110.bootpc: BOOTP/DHCP, Reply, length 300
16:41:59.543558 IP 10.10.10.1.bootps > 10.10.10.110.bootpc: BOOTP/DHCP, Reply, length 300
16:41:59.543727 IP 0.0.0.0.bootpc > 255.255.255.255.bootps: BOOTP/DHCP, Request from 3a:7f:58:51:00:fe (oui Unknown), length 288
16:41:59.555266 IP 10.10.10.1.bootps > 10.10.10.110.bootpc: BOOTP/DHCP, Reply, length 300
16:41:59.563481 ARP, Request who-has 10.10.10.110 tell 0.0.0.0, length 28
16:41:59.605850 ARP, Request who-has 10.10.10.110 tell 0.0.0.0, length 28
16:41:59.647475 ARP, Request who-has 10.10.10.110 tell 0.0.0.0, length 28
16:41:59.705023 IP 10.10.10.110 > igmp.mcast.net: igmp v3 report, 2 group record(s)
16:41:59.706246 ARP, Request who-has 10.10.10.110 tell 10.10.10.110, length 28
16:41:59.708714 ARP, Request who-has 10.10.10.1 tell 10.10.10.110, length 28
16:41:59.713860 ARP, Reply 10.10.10.1 is-at d8:3a:dd:2e:73:d7 (oui Unknown), length 28
16:41:59.713865 IP 10.10.10.110.60347 > 10.10.10.1.domain: 27152+ [1au] PTR? 110.10.10.10.in-addr.arpa. (54)
16:41:59.713871 IP 10.10.10.110.47141 > 10.10.10.1.domain: 60696+ [1au] A? fedoraproject.org. (46)
16:41:59.713872 IP 10.10.10.110.34846 > 10.10.10.1.domain: 9606+ [1au] AAAA? fedoraproject.org. (46)
16:41:59.713957 ARP, Reply 10.10.10.1 is-at d8:3a:dd:2e:73:d7 (oui Unknown), length 28
16:41:59.719409 IP 10.10.10.1.domain > 10.10.10.110.60347: 27152 NXDomain 0/0/1 (54)
16:41:59.724323 IP 10.10.10.1.domain > 10.10.10.110.47141: 60696 9/0/1 A 34.221.3.152, A 67.219.144.68, A 8.43.85.67, A 152.19.134.198, A 38.145.60.20, A 152.19.134.142, A 8.43.85.73, A 140.211.169.196, A 38.145.60.21 (190)
16:41:59.724668 IP 10.10.10.110.58064 > ec2-34-221-3-152.us-west-2.compute.amazonaws.com.http: Flags [S], seq 1848367379, win 64240, options [mss 1460,sackOK,TS val 1043404463 ecr 0,nop,wscale 7], length 0
16:41:59.752980 IP 10.10.10.110.mdns > mdns.mcast.net.mdns: 0*- [0q] 3/0/0 PTR Passim-7EB5._cache._tcp.local., PTR Passim-7EB5._cache._tcp.local., PTR _cache._tcp.local. (173)
16:41:59.774206 IP 10.10.10.110.mdns > mdns.mcast.net.mdns: 0 [4q] [6n] ANY (QM)? 1.1.2.d.3.5.2.1.6.7.1.b.d.a.f.1.0.0.0.0.0.0.0.0.0.0.0.0.0.8.e.f.ip6.arpa. ANY (QM)? fedora.local. ANY (QM)? 110.10.10.10.in-addr.arpa. ANY (QM)? Passim-7EB5._cache._tcp.local. (270)
16:41:59.783542 IP 10.10.10.110.44926 > 10.10.10.1.domain: 47816+ [1au] PTR? 22.0.0.224.in-addr.arpa. (52)
16:41:59.794242 IP 10.10.10.1.domain > 10.10.10.110.44926: 47816 1/0/1 PTR igmp.mcast.net. (80)
16:41:59.794716 IP 10.10.10.110.46580 > 10.10.10.1.domain: 51680+ [1au] PTR? 152.3.221.34.in-addr.arpa. (54)
16:41:59.795110 IP ec2-34-221-3-152.us-west-2.compute.amazonaws.com.http > 10.10.10.110.58064: Flags [S.], seq 4037310182, ack 1848367380, win 62643, options [mss 1460,sackOK,TS val 2589297896 ecr 1043404463,nop,wscale 7], length 0
16:41:59.795129 IP 10.10.10.110.58064 > ec2-34-221-3-152.us-west-2.compute.amazonaws.com.http: Flags [.], ack 1, win 502, options [nop,nop,TS val 1043404534 ecr 2589297896], length 0
16:41:59.795201 IP 10.10.10.110.58064 > ec2-34-221-3-152.us-west-2.compute.amazonaws.com.http: Flags [P.], seq 1:94, ack 1, win 502, options [nop,nop,TS val 1043404534 ecr 2589297896], length 93: HTTP: GET /static/hotspot.txt HTTP/1.1
16:41:59.799026 IP 10.10.10.110 > igmp.mcast.net: igmp v3 report, 2 group record(s)
[... cut ...]
16:43:10.726748 IP 10.10.10.110.35119 > time-h.den.codehof.net.ntp: NTPv4, Client, length 48
16:43:10.768629 IP time-h.den.codehof.net.ntp > 10.10.10.110.35119: NTPv4, Server, length 48
16:43:11.404271 IP 10.10.10.110.50298 > remote.frangipani.org.ntp: NTPv4, Client, length 48
16:43:11.428737 IP remote.frangipani.org.ntp > 10.10.10.110.50298: NTPv4, Server, length 48
16:43:11.606926 IP 10.10.10.110.34051 > ntp3.radio-sunshine.org.ntp: NTPv4, Client, length 48
16:43:11.701973 IP ntp3.radio-sunshine.org.ntp > 10.10.10.110.34051: NTPv4, Server, length 48

```

Any clues?

RabihND, 2024-11-01:
I use Ubuntu, so I configure the network manually with an **IP range from 10.0.0.0 to 10.0.0.255 and a subnet mask of 255.0.0.0. (needed for Raymarine as I see)**

You can try running the following temporary command to set up the network interface:

```bash
sudo ifconfig eth1:1 10.0.0.1 netmask 255.0.0.0 up
```

This involves adding an alias port, which allows a single physical ethernet card to have multiple IP addresses.

Note that `eth1` will vary with your hardware."

LisaRowell, 2024-11-04:
I've been working on getting a RD418D running as well, though my boat has one of the Raymarine Seatalk HS switches and I'm going through that instead of using a regular switch. With the Raymarine MFDs disconnected and an ethernet to my OpenCPN device configured with a static ip address of 10.18.x.x and a /16 netmask I'm able to connect and control the radar. The radar is showing up with the address of 10.18.65.32. This would seem to indicate that either the radar powers up with a fixed address (unlikely I would think), or the switch is running DHCP itself. 

Oddly if I configure my device to connect via DHCP to the switch I'm not given an address. It's possible I'm getting filtered based on hostname or mac address or something, but I haven't experimented with it that much. I noticed in the above logs that they're using an unregistered mac address block which in itself is a little odd.

For those having issues, have you been running a DHCP server in your setups?

Ideally I'd like to get this working over WiFi, but that's a work in progress...

opaezosuna, 2024-11-05:
> I've been working on getting a RD418D running as well, though my boat has one of the Raymarine Seatalk HS switches and I'm going through that instead of using a regular switch. With the Raymarine [...]

Thank you for sharing Lisa!
I did try with a DHCP server, no luck. Unfortunately for me, for some reason, my radar dome won't power up anymore so I will not be able to conduct new tests any time soon :(
Does the address of the radar change over time?
Do you have to turn on the MFD during a fresh power up to be able to connect from your OpenCPN computer?
I wanted to try sending a WOL message, but my radar did not power up anymore, so sad.

opaezosuna, 2024-11-14:
I just went through my notes and there is something I was also going to do eventually and it was to to look for UDP packages once the radar is connected tot the computer's Ethernet port. It may be the case that the network protocol is UDP and not TCP. But I'm no expert.

keesverruijt, 2025-11-10:
Reactivating this a little, as I am working on https://github.com/keesverruijt/mayara to include support for these older Ray radars, I was on a friends boat with Axiom+ and Quantum, it seems the ray marine stuff tries to get a DHCP address but if it finds there is no DHCP server then it falls back to a well-known-raymarine-range. This could be 10.18 for these radars.

Can you see what happens if you put a DHCP server on the network, either with a small router or on the computer that you are running OpenCPN on? On Linux and macOS this is trivially simple, Windows I don't know.