opencpn-radar-pi/radar_pi #264: Documentation/Extraction of Network Protocol for Simrad 4G Radar

avitase, 2024-07-12:
First of all, thank you for the fantastic work on the radar_pi plugin!

I have seen issue #260 where it was mentioned that running the plugin without OpenCPN is not currently possible. I am interested in exploring the possibility of creating a standalone solution for the Simrad 4G Radar. Specifically, my goal is to implement a translation layer for the radar data without incorporating other features such as the GUI.

Given that the Simrad 4G Radar is supported by the plugin, the necessary information about the network protocol is implicitly present in the source code. However, I am having difficulty locating the relevant parts of the code that detail the network protocol specifics.

Could anyone provide guidance on which files or sections of the source code specifically deal with the network protocol for the Simrad 4G Radar? Alternatively, if there is any existing documentation within the codebase that I might have missed, pointing me to those resources would be incredibly helpful.

Thanks!

Hakansv, 2024-07-12:
I'm not a developer but have a look at radar_pi\src\navico\NavicoReceive.cpp.
For general network connection see the the Navico part of the [manual](https://opencpn-manuals.github.io/main/radar/0.1/index.html)

For standalone purposes you may find some (old)basics [here](https://sourceforge.net/projects/openbr24/)

keesverruijt, 2024-08-14:
See https://github.com/keesverruijt/mayara

closed 2024-08-28 by keesverruijt