opencpn-radar-pi/radar_pi #268: Radar connection not found on eth0 Debian RPi5

Hakansv, 2024-08-17:
RPi5 Bookworm.
O 5.10
Radar plugin 5.5.4-beta
A radar scanner is connected to eth0. 
The eth0 Ipv4 is set to DHCP.
Radar plugin won't try to to use an IP but stay's on "Initiating".
Looking at Wireshark, see picture, the scanner is there but radar_pi don't scan eth0.
O Log say's:
11:15:21.193 MESSAGE NavicoReceive.h:111 Navico BR24 navico receive using addresses:  /236.6.7.8:6678/236.6.7.9:6679/236.6.7.10:6680

If I enable wifi on the RPi radar_pi start to scanning 192.168.1.87 (host's IP) interface but there's of course no scanner.

[image]

Hakansv, 2024-08-17:
On Win10 there's no problem to find and use the BR24.
There all networks, wired/wifi, are scanned until connection is established and the scanner can be used.

douwefokkema, 2024-08-17:
What do you get when you type ifconfig in the terminal?

Hakansv, 2024-08-17:
Douwe..

Maybe strange no IP address for eth0 although it's set to DHCP?

rpi5@rpi5:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether d8:3a:dd:a8:48:0f  txqueuelen 1000  (Ethernet)
        RX packets 3646  bytes 1155066 (1.1 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1119  bytes 333598 (325.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 106  

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.87  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 2001:2042:7535:bd00:5cf0:b920:9a82:3fdd  prefixlen 128  scopeid 0x0<global>
        inet6 2001:2042:7535:bd00:5ab2:9f71:96b4:dc60  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::3fa1:cf63:6895:4d75  prefixlen 64  scopeid 0x20<link>
        ether d8:3a:dd:a8:48:10  txqueuelen 1000  (Ethernet)
        RX packets 826431  bytes 122666105 (116.9 MiB)
        RX errors 0  dropped 38677  overruns 0  frame 0
        TX packets 914491  bytes 315756958 (301.1 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

keesverruijt, 2024-08-17:
If there is no IPv4 address there is no way it can send data. 

If you set it to DHCP but there is no DHCP server then apparently you don’t get an automatic IP address as on Windows. 

An automatic IP address (range 169.254/16) will work fine. 

Proof: give it any fake address, for instance:

     sudo ifconfig eth0 10.1.1.1 netmask 255.255.255.0 up

Hakansv, 2024-08-18:
@keesverruijt 
Yes, you're on the track there. But this would be new in Bookworm? In previous Debian this has worked out of the box. The radar plugin did always found the scanner without tweaks to the network card.

The ipv4 is set to DHCP but there is also an opportunity for an extra static address. I added "169.254/16" and then the scanner was found at once. But it was unresponsive. Couldn't show the Info window. I got it to transmit once but nothing more. Looked in the log and this message was flooding four to five times per second:
 02:38:57.166   ERROR NavicoControl.cpp:106 Navico BR24 Unable to transmit command: Network is unreachable
So he was all time busy to print to log.

Then in Rpi's network card ipv4 preferences I found a checkbox, see shot. When enabled all is fine and radar scanning works as normal and no log flooding.

I btw thought the Navico scanner would fix the DHCP assignment? But is that only for Windows as you mentioned? 

[image]

Hakansv, 2024-08-18:
This network is not stable.
Even if the radar transmits the log is flooding the above "Unable to..." message making O irresponsible. Looking at ifconfig no ipv4 address until I again save the network preferences or restart the scanner.
Could there be some timeout due to "no internet" in Bookworm?

Hakansv, 2024-08-18:
@leamas 
Do you know anything about changes in Bookworm about the network connections? 
I can get the ifconfig to report a ipv4 address by again saving network properties or restart the radar scanner but it won't last for long. Radar plugin again start to flood the log and ifconfig has no address.

keesverruijt, 2024-08-18:
If this is a "small" network with just your RPi and the radome, do not use DHCP: there is no DHCP server on that network, so this won't work. Just assign a fixed IP address. My guess is that the DHCP client is now removing your assigned address after X seconds.

Hakansv, 2024-08-18:
Changing eth0 to manual instead of DHCP make a difference, yes. It has now been running stable for an hour and connects nicely after a O restart. Also ifconfig is of course reporting a stable ipv4 address all time.
I do claim this is new for Bookworm? I've not been forced to change eth0 settings on previous Debian versions.
We may add to the manual that Bookworm need a manual address and not DHCP when directly connected without a switch? 
I'll also made some tests using a switch in between. Then I assume the DHCP would work as before.

My assumption that DHCP would work all time comes from the old base document. Apparently not valid for Bookworm?
Extract from "Roboat_DabrowskiEtAl2011"
> During boot-up the radar antenna transceiver and the display unit automati-
> cally configure their IP addresses. If no DHCP server is available, both devices
> select an IPv4 Zeroconf link local address as described in RFC3927[6]. All
> communication is done using UDP multicast datagrams. All units emit Inter-
> net Group Management Protocol (IGMPv3[7]) subscription announcements.
> This approach has two advantages. (1) It enables the system to handle multi-
> ple subscribers (i.e. controller/display units) and (2) The IP addresses of the
> devices are irrelevant, since they solely communicate through fixed multicast
> groups.

keesverruijt, 2024-08-18:
A switch is just a switch, it doesn't provide a DHCP server. If you'd add a wired connection to a router, then the router contains a DHCP server and it would work.

I have installed a Debian 12 ARM64 virtual machine and this has the same behaviour as you describe. I'm not sure this is really new though. I've seen it in Ubuntu 22.04 as well.

Thus, not say Debian 12 (bookworm) but "recent Linux editions". I will add this to the Wiki page.

keesverruijt, 2024-08-18:
I updated https://github.com/opencpn-radar-pi/radar_pi/wiki/Hardware-installation-(Navico)

Hakansv, 2024-08-18:
Yes a switch is a switch, I know. Fast typing.
Just to confirm I tested to connect the scanner to a router and a switch. When the eth0 properties was set to DHCP, and no additional static address, radar plugin found the scanner at 192.168.... as expected.

Your change to the wiki agreed. Very good!
Thanks

closed 2024-08-18 by Hakansv