opencpn-radar-pi/radar_pi #272: How can I connect Raymarine Quantum 2 Q24D to openCPN with Axiom?

li01233, 2024-12-16:
I purchased a Raynet to RJ45 cable, connected the cable to Quantum2 to the switch (the switch is working), and used Axiom's WIFI function to connect Quantum2 and start it. The current and power both indicate that the radar is running, and the radar data has been successfully plotted on Axiom. However, the OpenCPN radar plugin keeps showing that it is initializing Quantum. How can this be resolved? I also tried to connect Quantum2 directly to the computer, but it still displayed "initializing".

The radar_pi I used was found and installed in the OpenCPN plugin library, version v5.3.4, on Ubuntu 20.04, with IPv4 wired connection set to automatic (DHCP).

Should I also connect Axiom to the switch? Does the network cable not transmit data in WIFI mode? 
Do I need to buy a Raynet specific switch? Or do I need to connect to a router instead of a switch?

douwefokkema, 2024-12-16:
I have recently been working on the Quantum radar in radar_pi, I was able to test on the boat of a friend. I have tested the setup with the radar, Axiom and computer all connected to a (non managed Sitecom) switch. The radar was only recognized by radar_pi when the Axiom is online. It seems that the Axiom is having some interaction with the radar that I have not yet managed to decipher. But in this setup the radar works fine with the radar_pi on both Windows and Ubuntu. There were some bugs in some controls, these have been corrected in the current radar_pi master on GitHub (to be released). I did not test any WiFi connected setup. As the radar is said to be using unicast communications for WiFi instead of multicast, I do not expect that the radar_pi will function.
In your case however, I do not see how the computer with radar_pi is getting the radar messages when the switch is not connected to the Axiom or the radar. I would try to connect the Axiom to the switch as well. May be the Axiom will re-transmit the radar data that it receives though the WiFi. I just don't know. And for your info, the Axiom has a normally active DHCP server.

li01233, 2024-12-16:
Thank you for your reply. I will buy another Raynet to RJ45 cable to connect Axiom to the switch and try it out. By the way, can the radar data be recorded in radar_pi?

douwefokkema, 2024-12-16:
Buying a Raynet converter cable could be avoided by cutting a Raynet cable and putting RJ45 connectors on both sides of the cut. Google will find your pin layout, only 4 wires are important. 
Radar_pi can not record radar data. But radar data can easily be recorded with tcpdump (Linux) or wireshark (both Linux and Windows) into .pcap(ng) files. These files can be "replayed" with tcpreplay (Linus only)

douwefokkema, 2024-12-16:
John, you are right, once the communication between radar_pi and quantum radar has started, Axiom may be turned off. But in your case, when Axiom is unavailable, can you restart radar and radar_pi (from off) and get the communication going again? I managed only in some rare cases.

jpcavanagh, 2024-12-16:
Yes, I using a separate dhcp server I don’t need the axiom.  Note that my
quantum goes into ‘sleep mode' about 12-15 minutes after not getting an ip
address.  Raymarine suggest powering cycling the radar to make it work.

John

keesverruijt, 2024-12-17:
@douwefokkema can you add this to the documentation wiki pages?

douwefokkema, 2024-12-17:
@John, Yes, I also tried a DHCP server. This worked sometimes, but power cycling the radar will force refresh of IP addresses, good idea, thanks. In my DHCP server I used the addresses used by the Axiom server. What addresses did you use? May be there is no need to replicate the Axiom addresses as I did. I used the isc-dhcp-server with the following configuration:
#  /etc/dhcp/dhcpd.conf
default-lease-time 600;
max-lease-time 7200;
subnet 198.18.0.0 netmask 255.255.248.0 {
 range 198.18.6.10 198.18.6.200;
 option routers 192.168.0.254;
 }
But as said, may be this is unnecessarily complex. Any home router with a DHCP server could be enough in the default configuration. On top of that the Axiom DHCP server should be "off" or "automatic" but not "on"

jpcavanagh, 2024-12-17:
I used a generic class C address.  Something like 192.168.10.x  and I let
it allocate from 10 to 199.

I use a short lease time because when I close the VM running the DHCP
server I want the radar and cams to release their address before I turn on
the AXIOM.  I am not sure what might happen if the radar already has an
address once the AXIOM starts, I think it should eventually ask for a new
address and the AXIOM would serve.

Merry Xmas
John

douwefokkema, 2024-12-18:
@John, you could try to keep your own DHCP server always on and on the Axiom switch DHCP "off" or "automatic", I think these settings are in the diagnose menu of the Axiom.

li01233, 2024-12-24:
I can use tftp64+SDK on Windows without Axiom to get the data, but before I have to use its SDK to send the command to launch.
On Ubuntu, I still used the Axiom to connect to the switch and see the radar spokes in opencpn (provided that the Axiom sends the “launch” command first).

li01233, 2024-12-24:
But when using it I found another problem: I used his code  
[ROS-USV-Packages](https://github.com/Aquatic-Drone-Testbed/ROS-USV-Packages)
to read and visualize the radar data, and changed the code to ROS1 version in my repository.
[mul_modal_ROS](https://github.com/li01233/mul_modal_ROS/tree/master/src/radar)

But without opening the radar_pi in opencpn, the code will stay in a loop waiting for the data, and as soon as I open the radar_pi, the code will work fine. 
Is there any role for the radar_pi in this process (both my radar and Axiom are connected to the switch)?

keesverruijt, 2024-12-24:
> I can use tftp64+SDK on Windows without Axiom to get the data, but before I have to use its SDK to send the command to launch.

I don't know what this means, what SDK?

As to your last message, you're referring to code that we don't know at all. And then you say that you changed it, and that you don't understand why it doesn't work. 

And then you ask whether there is "any role for the radar_pi in this process". I don't understand what that means or what you are trying to ask.

closed 2025-01-10 by li01233