opencpn-radar-pi/radar_pi #275: Radar goes from Transmitting to Standby

henriberisha, 2025-03-29:
Firstly, big thanks to the developers for their great work and enabling us with this open project.

The following is not as much of an issue as it is an inquiry for information and better understanding of the radar functionality. 

I am part of a research project in which we have a Simrad Halo20+ installed on a vessel and one of the goals is to be able to use the radar remotely. 
The topology of the connection is:

**Simrad <-> IP Link Radio  <------> IP Link Radio <-> Host Computer**(Ubuntu 24.04.2) running OpenCPN (V5.10.2 and radar_pi 5.5.5)

I am having this phenomena happening in which the radar is transmitting and suddenly the PPI stops showing the image data and displays info about the radar and how many hours it has transmitted. Sometimes It keeps the same screen on PPI and other times goes into Standby mode and does not respond upon trying to start it. Other times i am able to restart the transmitting. I have noticed that this happens when the IP link of the radios starts to degrade. 

I have used wireshark to monitor real time the incoming packets, and i notice that even when the radar has just gone into non transmitting mode there are few incoming radar image data packets for a few seconds and they stop. My hypothesis is that the radios are queuing packets and maybe the radar or opencpn are having a timeout and one side closes the transmitting. Not very sure. 

I hope to get a possible answer by you. I look forward to one.
Thank you very much

Hakansv, 2025-03-29:
If I read correct you have connected the radar scanner by a wifi network?
That's normally not working unless the wireless settings are special, see the radar [manual](https://opencpn-manuals.github.io/main/radar/0.1/Hardware-installation-(Navico).html).
If the connection is not stable the "Keep alive" signal will be sometimes missed and the scanner goes to standby.
Try a wired network. Either direct or via a switch.

keesverruijt, 2025-03-29:
Radar communications over UDP multicast do not lend themselves to this setup, and it is not supported by this plugin.

I suggest you look at https://github.com/keesverruijt/mayara instead, which can be used like this:

Simrad > LAN > Mayara -> (wireless or other non deterministic) networking -> web browser / your application

henriberisha, 2025-03-29:
@Hakansv you are correct. The scanner is being used over a wireless network over long distances and not a home-like wireless LAN. Thank you for the tip, probably it is missing the "Keep Alive" signal. I will look more into the code to find it. If it is possible, would you be able to point me where in the code the Keep Alive transmission happen? Thank you.
is it that the Scanner stops the transmission itself, or opencpn tells it to stop?

@keesverruijt thank you for your suggestion, i will look at this application. Will "mayara" enable me to get information about radar blobs? For example if i want to create a JSON message with relevant blob data, will it be possible by doing API calls via mayara?

Thank you very much

keesverruijt, 2025-03-29:
Please just look at mayara first and then inquire on the issues page there. Let's not clog up discussions here with a different project.

Hakansv, 2025-03-29:
re: " probably it is missing the "Keep Alive" signal. I"
No i didn't mean the signal is missing from the radar plugin. It's sent by the plugins as soon "Transmit" is selected. It's received by the scanner. If not received in due time the scanner goes to standby. What I tried to express is if the (wifi)network is busy the keep alive is not transferred in due time and the scanner goes down. That's one of the reason a wifi network can make trouble.

henriberisha, 2025-03-29:
@keesverruijt @Hakansv thank you very much for the info. Super helpful.

One more question and will close the thread, for a Halo 20+ is it better to select Navico 4G A/B or the Halo A/B? 

Respects

Hakansv, 2025-03-31:
The users of 4G or Halo might have an answer. I may recall some discussion about some (old?) Halo is actually a 4G but no details in my mind. I myself have a 3G.

keesverruijt, 2025-03-31:
Halo, because otherwise you don’t have Doppler.

henriberisha, 2025-04-01:
Thank you very much for the information! I appreciate your answers

closed 2025-04-01 by henriberisha