opencpn-radar-pi/radar_pi #276: Potential position issue

redhog, 2025-04-11:
While working on https://github.com/KAHU-radar/radarhub-opencpn I noticed that SetPositionFixEx is not called for every GGA/GLL/RMC message, but much more seldomly.

In my code, I calculate lat/lon for an ARPA target based on own boat la/lon and bearing/distance and construct a trace. This lack of position update meant that my trace was jumping back and forth, esp. at higher SOG!

I'm pretty sure you're doing something very similar in radar_pi for radar overlay (and AIS target matching?), as you do use SetPositionFixEx.

douwefokkema, 2025-04-12:
radar_pi is indeed using SetPositionFixEx  for position updates, but these updates appear about every second. Therefor radar_pi also is also using received nmea083 messages directly through SetNMEASentence(wxString &sentence). As the target tracking is very sensitive to inaccurate positions, I added a Kalman filter (GPSKalman) to predict and smooth out the received positions. This improves the accuracy of the positions at any given time.

redhog, 2025-04-13:
Ah thanks! Nice point about Kalman filter, will add the same to radarhub (alternatively, if radar_pi emitted RATTL instread of RATTM I could rely on your code, but I guess other things would break due to such a change).