opencpn-radar-pi/radar_pi #277: NMEA for ARPA/AIS match

redhog, 2025-04-11:
When an ARPA target matches an AIS target by position, it is marked as L (Lost) in the $RATTM so that the two targets don't overlap on screen. However, it might under some circumstances be useful for other external tools to be able to distinguish an actual lost target, and an AIS-overlaid target, that is, to know that we're actually seeing this target on radar, as well as which ARPA target corresponds to which mmsi.

Would you be open to a PR that emits custom NMEA sentences that convey this information, and would you have some feeback on how these should be structured?

douwefokkema, 2025-04-12:
In the current radar_pi AIS targets and radar targets are different objects from two different origins. And a radar target at the location of an AIS target is suppressed. In real world however there are only targets (objects around us) that we try to follow. So a radar target and an AIS target is the same thing, observed by different means. So we should follow targets, and improve the information we have on that target with both radar and AIS. I am currently working on a version of radar_pi that handles targets in this way. 
This new approach has more implications than I expected, and takes more time. In the current radar_pi targets exist per radar. In the new approach targets exist independent of the radar that follows them, the target may be seen by any radar(s) and/or the AIS. If a target is seen (also) by AIS it will have the mmsi as identification. In case the AIS signal disappears, it will still be a radar target with the same mmsi.

Coming back to your question: Would you be open to a PR that emits custom NMEA sentences that convey this information, and would you have some feedback on how these should be structured?

Answer is no, that would no longer be needed as there will be no essential difference between radar and AIS targets anymore.

redhog, 2025-04-13:
Well, I would counter that in your new definition of a target, this coordination would be even more important - as you'd need to emit some kind of messages for targets with different known information about them - and tie their identities together. As NMEA RA* messages don't do mmsi, and AIVDM don't do non-mmsi vessels, you'd need to add something else.

That is, if you still want consumers of the NMEA stream from opencpn to be able to use the complete information your plugin generates?

In radarhub, I'm using the output of your plugin to crowdsource radar detections/paths of vessels that either do not have AIS or have turned AIS off, in the latter case correlating with AIS tracks to eventually establish an mmsi for the full track.

The goal is to provide better information on activities that affect the marine ecosystem.

I'm hoping that your plugin will continue to be possible to use this way in conjunction with my plugin?