opencpn-radar-pi/radar_pi #278: New radar library :)

redhog, 2025-05-05:
Heyas!

This isn't an actual issue, it's justa FYI:

I've been developing a binding for the radar classes from radar_pi for NodeJS, so that they can run outside of this plugin for interactive/real-time experimentation (both with the code, and with received data). The node_js binding module is entirely headless (but still needs to link to wxWindows to make wxThreads etc work).

https://github.com/KAHU-radar/radar_receiver

The structure of my code is that radar_pi is pulled in as a git submodule. All the radar-specific include and src subdirectories are used as-is. Several other header files and some source files are to.

RadarInfo and radar_pi are entirely replaced with API-compatible classes. RadarControlItem is replaces with a class with the same functionality but also the ability to register a callback that listens for value changes.

I will try to get Arpa to work too, but that seemed a bit harder as the code is more intertwined with GUI controls. If I can make that work, this could be the start of a useful SignalK plugin...

The radar_receiver library obviously has the same license as radar_pi :)

Looking forward to your thoughts on this and any feedback on my code!

keesverruijt, 2025-05-05:
Oh, you may have missed that we are moving in the other direction: reimplementing the plugin in a standalone server that is easy to integrate with.

See https://github.com/keesverruijt/mayara

You can talk to the mayara server using any language that can talk web sockets and protobuf (binary data with the actual spokes.) From nodejs this would be a no-brainer. 

There is a radar channel on the Signal K discord server that you may also want to participate in: https://discord.gg/C9chAVMh

There are two implementations of the provisional "Signal K radar API", my Mayara project (written in Rust) and https://github.com/wdantuma/signalk-radar written in Go. Willem isn't moving so fast anymore but Mayara is steadily progressing. Navico work is 99% complete, with all controls. Raymarine and Garmin have stubs. Somewhat further along is Furuno, with working decoding of spokes but hardly any controls yet and many issues still. Target trails work. ARPA is work in progress.

The Mayara server compiles to a standalone server with zero dependencies running on medium level arm64 routers (generally the ones capable of WiFi 6) and anything bigger than that.

redhog, 2025-05-06:
Oh that is cool! That is definitely the way to go. But, as a stand-alone server (as opposed to a library), how does the installation look like on signalk? I assume then it involves more than just installing a plugin from the GUI?

The radar support is a bit of a worry for me: radar_pi doesn't have Furuno, but Mayara does have Furuno, at the same time, radar_pi supports a whole bunch of radars that Mayara does not yet?

I chose my approach to try to make sure that /all/ the radars currently supported by radar_pi would still work. What is the API inside Mayara for a radar driver? I'm wondering if we could use my approach to "wrap" existing radar_pi radar drivers to work inside Mayara temporarily until they have all been properly ported?

I'm pretty convinced I can make my build work without wxWidgets too - the subset of wxWidgets stuff used in the radar drivers is pretty small (mostly threading and networking), and should be poossible to stub out.

keesverruijt, 2025-05-06:
Building a radar server in Node.js is doomed to fail as soon as you try to do CPU intensive stuff like ARPA. Also Signal K !== signal k reference server in node.js.

Installation is simple; just drop the binary in and start it. Willem has written a small plugin for Signal K to connect it up.

What is still being worked on is support in charting apps like Freeboard SK. In my opinion that is out-of-scope for the project itself: it is a server, anybody is free to interact with it. I include a demo PPI application that shows purely the radar data. Willem has modified Freeboard SK so it shows radar as well; see our docker image demos in the respective demo directories.

Obviously I intend to re-implement all radars in Mayara. I don't have access to a Garmin radar yet. I will have almost full time remote access to a Quantum 2 soon. So Garmin will get in but untested by me. Note that the Garmin support is for radars that are getting quite old, so hopefully someone with a recent Garmin steps up.

Adding a radar in Mayara is just as easy as in radar_pi, maybe slightly easier. Rust is so much more better behaved making it easier to understand. I'd love for you or others to start helping me!

redhog, 2025-05-06:
So there are two things I'm interested in in nodejs: ARPA done in rust/c++ sending targets to nodejs, and having access to spokes and typed arrays sent to other non-javascript code, via javascript. In particular, sending spokes to the web client in a format that can be deserialized to typed arrays w/o javascript loops and sent directly into webgl (if you haven't already, you might want to look into what serialization formats have fast JS decoders that decode into typed arrays).

Having to install binaries outside of the plugin system means a lot of "sysadmin overhead" - esp. in device the user typically only access via the web, such as the Victron GX (or what we're building). Writing c++ or rust code against napi and orchestrating it from javascript, means you get to reuse npm for deployment, and any GUI that uses npm (such as the signalk plugin gui) then gets this for free. How hard would it be to add this as a compile target / alternative "main" for mayara?

Where would it be most useful for me to start to help you?

I was thinking that the most obvious point would be to wrap the existing radar_pi code as a "mayara radar" class to be used until native implementations are available, since I'm kinda halfway on that anyway? But maybe there are better starting points?

keesverruijt, 2025-05-06:
Please DM me (kees.verruijt) on Discord

keesverruijt, 2025-08-01:
@redhog has since stopped working at KaHu and this is now a dead end.

closed 2025-08-01 by keesverruijt