opencpn-radar-pi/radar_pi #281: Issues with connecting to Raymarine Quantum Q24C

LasseGH22, 2025-06-20:
I am a software engineer student with little to none experience with networking which i think is the cause of my issue.

I have had several issues trying to get this plugin working with my Raymarine Quantum Q24C. I am currently working in Ubuntu 22.04 with the radar connected to a normal tp-link router and my laptop connected to the same.

1. The link under "Installing the plugin" in the software installation wiki page results in 404 page not found.
I have now downloaded what i believe to be the same plugin from the GUI of OpenCPN. 
I found that when downloading OpenCPN via the apt repository the plugin GUI does not work as intended. When pressing the "Update Plugin Catalog: master" button nothing updates. Downloading it via flatpak instead fixed the issue for me.

2. Now that the plugin is installed the radar display keeps circling between 4 private addresses: "192.168.x.x", "10.126.x.x", "172.17.x.x", "172.18.x.x".  Listening for UDP packages via a tcpdump command on my laptops networking card shows packages being recieved from the private ip adress given to the radar by my router.

3. I have tried letting OpenCPN through my firewall without success which has led me to disable it fully, also without success.

4. I have tried getting it to work using Mayara (Without reading the README saying there is no support for Raymarine Quantum) which was able to discover my radar on this address: 232.1.143.1:2574

As mentioned i dont have much expertise in networking so i am completely lost as to what the next step could be.

jpcavanagh, 2025-06-20:
Point 2 implies that there is no network connection.

Ensure that you have a physical connection by checking that the radar and
pc running opencpn are plugged in to the same Ethernet switch.

After that it could be an addressing issue.  Do you have a dhcp server on
the Ethernet that can give both the radar and pc an op address?   Typically
this will be an MFD such as an Axiom.

You might check this by looking to see if your pc has been given an ip
address by the network.  ‘ip a’ command can help.

If you don’t have an mfd or it is powered off then you will need to create
a dhcp server on the Ethernet.

LasseGH22, 2025-06-23:
I have my laptop and radar connected via a tp-link router. I am pretty sure that my router has a built in DHCP server, but feel free to correct me if i am wrong in thinking it will do the same. From my routers admin panel i could see that my laptop had 192.168.0.101 and the only other device connected (Which i assume must be the radar) had 192.168.0.100.

You are right to assume that i am not using an MFD and it is also not my plan to do so

keesverruijt, 2025-11-19:
@Krecik7 hijacking issues is not normal on github.

If you want a reaction you will need to open a new issue.