opencpn-radar-pi/radar_pi #282: little problem to build with cmake version 4

filochard, 2025-06-25:
Hi
You may perhaps know that I am still building rpms of OpenCPN and its plugins for a linux distribution...
This allowed recently to find some problems to compile with the last version of GCC (15.0)
That have been corrected for the next version of OpenCPN (opencpn-libs/api-20/)

Now we use cmake version 4

That allowed me to find a strange error : radar_pi is apparently using opencpn-libs @ 9804fbe
 from https://github.com/leamas/opencpn-libs/tree/main
and inside it using  opencpn-libs/api-16/  instead of opencpn-libs/api-18/

We get this error : 
CMake Error at opencpn-libs/api-16/CMakeLists.txt:10 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

inside opencpn-libs/api-16/CMakeLists.txt  
and inside opencpn-libs/api-18/CMakeLists.txt   
there's 
cmake_minimum_required(VERSION 3.1)

this should be now 
cmake_minimum_required(VERSION 3.5)

To build we needed to use this workaround : 
%cmake -DCMAKE_POLICY_VERSION_MINIMUM=3.5

Nevertheless this should be corrected by using a more recent commit of opencpn-libs such as 433aed02b6dbf1b34709e44aea038f5c853bec79
from https://github.com/OpenCPN/opencpn-libs/

filochard, 2025-06-29:
Hi again
I could build with cmake 4.0 using a more recent commit of opencpn-libs : 
opencpn-libs-facd6f684e8b851619a100e5601626e2002e1d42   ....
all its CMakeLists.txt contains   cmake_minimum_required(VERSION 3.5)

So 
I didn't have to add : 
 -DCMAKE_POLICY_VERSION_MINIMUM=3.5 
at the end of  %cmake line of my spec file

keesverruijt, 2025-08-01:
I would not know who you are and certainly not that you release .rpms. Is that well known somewhere? 

I updated the ship driver build files to sd3.3.1, please check again. 

For all plugins that have an `update-templates` file in their root, they re-use the build files originating from Shipdriver plugin.

All I needed to do was:

    bash ./update-templates sd3.3.1

to get the build working again with new CMake. This has now been pushed to this repo.

filochard, 2025-08-01:
> I would not know who you are and certainly not that you release .rpms. Is that well known somewhere?
>
> I updated the ship driver build files to sd3.3.1, please check again.
>
> For all plugins that have [...]

Hi
I have been building rpms of OpenCPN and its plugins for Mageia for years when it was easy to do this.
Since the developers have promoted flatpak for Linux distributions and have suppressed the simple way to build packages I have had to add patches to the sources to continue ....
Since I build inside a local build system (mock) for the next version of our distribution, I have found problems with cmake4 and GCC15 ... I created patches to workaround this problems and I have communicated with the main developers to inform them of the problem they had not yet encountered (not using cmake4 or GGC15) before OpenCPN 5.12.0 is published

You may find my posts in the issues or discussion of OpenCPN itself

In the bottom of the download page you will find a link to Mageia and a list of the rpms done for this distribution
https://opencpn.org/OpenCPN/info/downloadopencpn.html

They are fully tested before being provided to Mageia users

filochard, 2025-08-01:
@keesverruijt 
Hi again
I have just seen the work you are doing on the radar code source : I hope this will work for flatpak I will test it as soon it is provided by the plugin manager...
On my side I have chosen to continue to build working rpms (with updated version of opencpn-libs) and built on the same system basis to prevent bad surprises...

filochard, 2025-08-04:
@keesverruijt 
Hi
I don't know if this will be useful for you
I continue to create rpms of OpnCPN and its plugins for Mageia
I have created a rpm of OpenCPN-5.12.2 and installed it as an update of OpenCPN-5.12.0 (which was already an update of 5.10.2)

I had modified the source of radar_pi 5.5.5 to build a rpm upon a more recent version of opencpn-libs it works with OpenCPN-5.12.2 it works !

I have now built radar_pi with a clone of your last commit (57abc72) it works perfectly on my system !
You might tag it as radar_pi 5.5.6 or 5.6.0 so that this is easier to download...
And maybe this should work inside flatpak 

Nota Bene 
Radar is no more proposed by the catalogue inside the plugin manager for the flatpak version of OpenCPN, even if the catalogue is updated

keesverruijt, 2026-02-11:
We don't use the opencpn-libs tree ourselves, as all CI builds download opencpn-libs automatically during build. However, I see how this can be useful for downstream projects.