opencpn-radar-pi/radar_pi #283: Plugin in disappears from latest update 5.12.0

carryonsailing, 2025-07-29:
I have tried on multiple fedora machines and the plugin disappears completely. Therefore I downgraded to older.
I am sorry if I am writing worthless stuff that you all know. Just trying to express my support and gratitude as a noob. 
Cheers you heros.

filochard, 2025-08-07:
> Guys I just want to express my gratitude for your efforts on the plugin and let me know if I can be of any service in that regard. I have mac and fedora machines and a home lab for testing if you need me. Kind regards (James Carry on sailing).

Hi 
There's an other issue where we talk about radar_pi there :  #282
[keesverruijt](https://github.com/keesverruijt) is working on the code

carryonsailing, 2025-08-10:
Just tested on two fedora machines, Radarpi Now working as Beta plugin. Incredible work, thank you.

closed 2025-08-10 by carryonsailing

filochard, 2025-08-10:
> Just tested on two fedora machines, Radarpi Now working as Beta plugin. Incredible work, thank you.

Hi 
I don't understand what kind of test you do on your fedora machines : 
do you use OpenCPN inside flatpak ?
or have you installed rpms of OpenCPN specially built for Fedora or imported from Mageia ?