opencpn-radar-pi/radar_pi #285: a little link problem to opencpn-libs

filochard, 2025-08-13:
NOTA BENE : I'm not sure that this is pertinent , nevertheless I create this issue that I may close if it is useless

Actually the radar_pi code is linked to an old version of opencpn-libs that has not been updated for 2 years from
https://github.com/leamas/opencpn-libs

It should be linked to 
https://github.com/opencpn/opencpn-libs

inside it you may choose a recent commit including an api-20 folder to be compliant with the last version of OpenCPN 5.12.2

Besides this to build radar_pi 
 I have modified line 188 in /Plugin.cmake
set(PKG_API_LIB api-16) # A directory in libs/ e. g., api-17 or api-16
into
set(PKG_API_LIB api-20) # A directory in libs/ e. g., api-17 or api-16

keesverruijt, 2025-08-13:
Why would we claim that we need API v20 when we haven't changed our requirements and no testing has been done with this API? I have not looked at the differences between them but just increasing the minimal API is not something to be done this lightly.

filochard, 2025-08-13:
> Why would we claim that we need API v20 when we haven't changed our requirements and no testing has been done with this API? I have not looked at the differences between them but just increasing the minimal API is not something to be done this lightly.

inside https://github.com/opencpn/opencpn-libs
the successive commits of opencpn-libs are connected with the versions of OpenCPN... the folder /api-20/ appeared with 
OpenCPN 15.2.1

In the last versions of opencpn-libs contains an /api-20/ folder (created upon OpenCPN 15.2.1 ) in which /api-20/ocpn_plugin.h contains  `#include <cstdint> `which allows to build with GCC15 (this doesn't exist neither inside api-16 nor api-18)

You may not know that I continue to build rpms for Mageia

For what I have seen, building locally OpenCPN 15.2.2 itself and its plugins, I discovered the need of : 
cmake_minimum_required(VERSION 3.5) in the CMakeLists.txt files to use cmake4
and the need to add  `#include <cstdint> `  for GCC15 inside ocpn_plugin.h
I have patched locally the sources to allow these builds

I have informed Alec Leamas and Dave Register when I discovered that and they modify the source !
 (this will be needed for the next version of Debian)