opencpn-radar-pi/radar_pi #287: Display problem with dual O system

JohnGC, 2025-09-26:
I have setup with two O plotters.
Halo 20+
Netgear switch
Yacht Devices SeaTalkng to Ethernet converter and DCHP server.
2 x BeeBox running Win 10 (with different resolution displays).
No MDF.

O version is 5.12.2.0 (both systems)
RADAR plugin version 5.5.5 (both systems)

Chart overlay mode for the RADAR.

Using RADAR with only 1 PC, no problems.

Using the RADAR plugin with both PCs simultaneously has the following problem.

On the second PC to access the RADAR, the image is painted briefly. Disappearing within less than 1/2s. Then isn't painted at all for a few degrees. Then repeat.

At some chart zoom levels it works correctly, but not always.

https://youtu.be/RPpzXjhDNWw

Short video of the issue.

douwefokkema, 2025-09-28:
Thanks for the logs. In your case both radars have the range on Automatic. This means that the radar_pi adjusts the radar range to match the chart scale. As both radars have different chart scales the radars are competing to set the radar range. I expect the first one to win. Solution: set radar range to a manually selected range. This range will then apply to both systems. Alternatively, use A radar on one system and B radar on the other. Then you can select the radar ranges independent from each other.

JohnGC, 2025-09-28:
Thanks for the very quick response.

I can confirm that setting one system to manual, removes the issue.

I'll experiment with the A/B option in the next few days.

Regards,

John

closed 2025-09-28 by douwefokkema